freifunk-berlin/firmware #194: release testing procedures

booo, 2015-01-02:
It's the second time I would like to push a release and we don't have agreed on some testing procedures. We should definitly change this! Some sketches for a testing szenario follow:

Internet <--> VPN03 <--OPENVPN--> gateway0 <--mesh (OLSR) --> node0

Configuration for the gateway:

* Password: foo
* Community: Freifunk Berlin
* hostname: gateway0
* ...
* Internet teile (check)
* special cert
* IPs
** 10.0.0.1
** 10.0.0.2
** 10.0.1.0/24

Configuration for the node:

* Password: foo
* Community: Freifunk Berlin
* hostname: node0
* ...
* Internet teile (check)
* special cert
* IPs
** 10.1.0.1
** 10.1.0.2
** 10.1.1.0/24

Tests:
* are all interfaces up (ip link show)
* do all interfaces have IPs (ip address show)
* do we have mesh neighbours (neigh.sh)
* can we ping our neighbours
* is openvpn running on the gateway
* can the gateway ping the internet e.g. vpn server
* does the node create a ipip-tunnel to the gateway
* do we have internet as node

I suggest we implement some automation with phantomjs or something the like and create some shell script that assist in testing stuff on the terminal level.

SvenRoederer, 2019-03-02:
https://github.com/freifunk-berlin/firmware/issues/667 also likes to have better testing-procedures

pmelange, 2019-03-02:
Running the wizard, and making sure that the values in /etc/config are set properly, would also be an important test.  This includes pulling in the defaults from /etc/config/freifunk and overriding them in the selected community profile (see #617).

xa-y-ax, 2020-05-05:
https://wiki.freifunk.net/Berlin:Firmware:Testing

SvenRoederer, 2020-05-05:
I'd like also to mention https://github.com/freifunk-gluon/gluon-qemu-testlab (python scripts to run qemu and gluon based virtual mesh networks) - This might be a framework, that we can adapt to our needs or get some inspirations. An automated setup to test the network configuration (Firewall, policyrouting, basic IP-configuration) is an repetitive task which  should be automated.

@kruszksa also mentioned the he once used something docker based for automated testing.