freifunk-berlin/firmware #198: [luci] "Speichern" vs "Speichern und Anwenden"

mxm, 2015-01-11:
In the administration settings, there is a "save" and a "save & apply" button. To me, this does not make any sense. Why would you change settings but not apply them?

If there was an actual flag indicating whether the saved settings have been applied or not this could make sense. However, currently, it is indistinguishably whether settings have been saved & applied or merely saved.

mxm, 2015-01-11:
I just realized, there is an actual flag in the right corner which lists all changes if you click on it. You can then also apply the settings. If that's the way, luci works, then probably we can't change much about this behavior.

booo, 2015-01-20:
"save" saves the configuration in /etc/config/*. "Save & apply" saves the configuration and e.g. restarts the services that depend on the changed configuration file.

I close this one because I think we can't do much about it.

closed 2015-01-20 by booo