freifunk-berlin/firmware #199: Services via OLSR announcen geht nicht mehr

faust2k, 2015-01-13:
Es sieht so aus als könne man keine Services mehr announcen - oder hat sich da was im Syntax geändert?
```
config LoadPlugin
        option library 'olsrd_nameservice.so.0.3'
        option suffix '.olsr'
        option hosts_file '/tmp/hosts/olsr'
        option latlon_file '/var/run/latlon.js'
        option services_file '/var/etc/services.olsr'
        list hosts '10.31.7.1 test'
        list service 'http://test.olsr/|tcp|Test'
```

```
root@f2a-core-rt:~# grep test /tmp/etc/olsrd.conf 
        PlParam "10.31.7.1" "test"
        PlParam "service" "http://test.olsr/|tcp|Test"
root@f2a-core-rt:~#
```

Die Liste ist mittlerweile auch ziemlich leer geworden. Ich glaube alle Services die da noch zu sehen sind werden von Routern mit der pberg-Firmware announced:
```
root@f2a-core-rt:~# cat /tmp/etc/services.olsr
### this file is overwritten regularly by olsrd
### do not edit

http://104.8.1.60:8000/stream.html|tcp|FalkenCam No.1 Eliaskirche               #10.31.3.226
http://104.8.1.60:8001/stream.html|tcp|FalkenCam No.2 Eliaskirche               #10.31.3.226
http://fluxfm-core.olsr:8000/fluxfm-berlin.mp3|tcp|FluxFM Berlin / MP3          #10.31.1.69
http://fluxfm-core.olsr:8000/fluxfm-berlin.ogg|tcp|FluxFM Berlin / OGG          #10.31.1.69
http://10.36.5.188:80|tcp|Luftraum Berlin               #10.31.0.61

### written by olsrd at Tue Jan 13 18:59:53 2015
```

booo, 2015-01-25:
Angeblich funktioniert die folgende Konfiguration:
```
config LoadPlugin
        option library 'olsrd_nameservice.so.0.3'
        option suffix '.olsr'
        option hosts_file '/tmp/hosts/olsr'
        option latlon_file '/var/run/latlon.js'
        option services_file '/var/etc/services.olsr'
        option ignore '0'
        list service 'http://10.36.77.10:80|tcp|ServNix77'
```

Bist du sicher, dass es nicht funktioniert?

booo, 2015-01-26:
Working configuration on liegewiese.olsr (kathleen):
```
config LoadPlugin
        option library 'olsrd_nameservice.so.0.3'
        option suffix '.olsr'
        option hosts_file '/tmp/hosts/olsr'
        option latlon_file '/var/run/latlon.js'
        option services_file '/var/etc/services.olsr'
        list service 'http://10.230.0.65:80|tcp|foo' # works
        list service 'http://liegewiese.olsr:80|tcp|bar' # works
        list service 'http://liegewiese.olsr:80/|tcp|baz' # works
```

I'm closing this as invalid.

closed 2015-01-26 by booo