freifunk-berlin/firmware #200: SSH über IPv6 Link-local partiell kaputt?

faust2k, 2015-01-13:
Ich kann das Problem derzeit nur auf der Segenskirche reproduzieren. Hier der Versuch sich zwischen segen-core (Kathleen) und segen-nw-5 (Kathleen) zu verbinden:

```
root@segen-core:~# ssh fe80::de9f:dbff:fe95:e60%eth0.100
Segmentation fault
root@segen-core:~# 
```

Logread/dmesg sagt dazu:
```
[ 3317.609760] ssh[7522]: unhandled signal 11 at 0000000d nip 480a7d1c lr 480a0f38 code 30001
```

Und logread auf segen-nw-5:
```
Tue Jan  6 11:22:57 2015 authpriv.info dropbear[2719]: Child connection from fe80::c24a:ff:fe51:2cd%br-fflan:41122
Tue Jan  6 11:22:58 2015 authpriv.info dropbear[2719]: Exit before auth: Error reading: Connection reset by peer
```

Als auf segen-nw-5 noch die pberg-Firmware lief ging das problemlos. Das gleiche gilt auch für segen-nw-2, segen-sw-5 und segen-sw-2 nachdem diese auf Kathleen migriert wurden.

Irgendwelche Ideen?

cholin, 2015-01-29:
Hast du hier noch weiter gedebugged? Konntest du es evtl eingrenzen? Welche Hardware ist konkret im Einsatz? Wie sind die Geräte miteinander verbunden (via OLSR mit VLAN?)? Benutzt du die stable Kathleen?

cholin, 2015-02-12:
Ich würde das Issue erstmal schließen, sofern @faust2k nicht mehr Informationen wie zBsp `ssh -vvv` hier pasted.

cholin, 2015-02-24:
Nicht reproduzierbar. Schließe das Ticket und tagge es unter `unresolved`.

closed 2015-02-24 by cholin