freifunk-berlin/firmware #201: [Usability] Anordnung der Buttons im Freifunk Wizard

faust2k, 2015-01-13:
Weil ich gerade wieder darüber gestolpert bin:

Es ist nur ne Kleinigkeit, aber aus der Usability-Perspektive durchaus relevant: der 'Next'-Button im Assitenten sollte rechts angeordnet werden, der 'Cancel'- oder 'Back'-Button links. Zumindest solang wir kein arabisches Release rausbringen... ;-)

http://uxmovement.com/buttons/why-ok-buttons-in-dialog-boxes-work-best-on-the-right/

cholin, 2015-01-29:
@geirkairam was ist deine Meinung dazu? Finde auch dass es sich um eine Verbesserung handelt. Magst du das vlt schnell umsetzen?

geirkairam, 2015-02-02:
ich hab das schonmal gehört, dass menschen eher af den cancel button drücken als auf den weiter butten :/ leider weiß ich nicht wie ich das schnell mal ändern kann.
die buttons sind so durch luci angeorndet und ich hatte nur die [beschriftung geändert](https://github.com/freifunk-berlin/firmware-packages/blob/master/utils/luci-app-ffwizard-berlin/luasrc/model/cbi/freifunk/assistent/shareInternet.lua#L7).

booo, 2015-02-03:
Maybe it's as easy as swapping some blocks in the template? We could create a local patch and/or try to push this upstream.

```diff
diff --git a/modules/luci-base/luasrc/view/cbi/simpleform.htm b/modules/luci-base/luasrc/view/cbi/simpleform.htm
index 5216cd5..66ec525 100644
--- a/modules/luci-base/luasrc/view/cbi/simpleform.htm
+++ b/modules/luci-base/luasrc/view/cbi/simpleform.htm
@@ -36,11 +36,6 @@
 <%- if self.flow and self.flow.skip then %>
        <input class="cbi-button cbi-button-skip" type="submit" name="cbi.skip" value="<%:Skip%>" />
 <% end %>
-<%- if self.submit ~= false then %>
-       <input class="cbi-button cbi-button-save" type="submit" value="
-               <%- if not self.submit then -%><%-:Submit-%><%-else-%><%=self.submit%><%end-%>
-       " />
-<% end %>
 <%- if self.reset ~= false then %>
        <input class="cbi-button cbi-button-reset" type="reset" value="
                <%- if not self.reset then -%><%-:Reset-%><%-else-%><%=self.reset%><%end-%>
@@ -51,6 +46,11 @@
                <%- if not self.cancel then -%><%-:Cancel-%><%-else-%><%=self.cancel%><%end-%>
        " />
 <% end %>
+<%- if self.submit ~= false then %>
+       <input class="cbi-button cbi-button-save" type="submit" value="
+               <%- if not self.submit then -%><%-:Submit-%><%-else-%><%=self.submit%><%end-%>
+       " />
+<% end %>
                <script type="text/javascript">cbi_d_update();</script>
        </div>
 </form>
```