freifunk-berlin/firmware #202: Missing AP traffic-stats in collectd

basros, 2015-01-17:
Traffic statistics are only shown for the following interfaces:
* ffvpn (if VPN is configured via wizzard)
* br-dhcp 
* wlan*-adhoc

```
# uci show luci_statistics.collectd_interface.Interfaces
luci_statistics.collectd_interface.Interfaces=br-dhcp wlan0-adhoc-2
# uci show luci_statistics.collectd_iwinfo.Interfaces
luci_statistics.collectd_iwinfo.Interfaces=wlan0-adhoc-2 wlan0-dhcp-2
```

There are iwinfo-stats for every wifi-interface, so I guess there should also be entries for wlan-dhcp interfaces in traffic-stats.

booo, 2015-01-20:
We could try to add them here:

https://github.com/freifunk-berlin/firmware-packages/blob/master/utils/freifunk-berlin-statistics-defaults/uci-defaults/freifunk-berlin-statistics-defaults#L35

Or add them in the assistent:

https://github.com/freifunk-berlin/firmware-packages/blob/master/utils/luci-app-ffwizard-berlin/luasrc/model/cbi/freifunk/assistent/wireless.lua#L186

I suggest that we create a patch for the assistent.

booo, 2015-04-30:
The master branch should contain a fix.

closed 2015-04-30 by booo