freifunk-berlin/firmware #206: luci-app-owm leaks potentially private information

sarumpaet, 2015-01-23:
luci-app-owm pushes information about all the router's network interfaces to the map api. This is somewhat surprising and not a good practice given that the Assistent allows to set up a private wifi network (that might use a hidden ESSID, etc.). Perhaps limit to **wifi** interfaces in **FF** zones that are **unencrypted**? That would also remove some clutter in the map db I guess.

booo, 2015-01-23:
In the long run we would like to migrate to the **libremesh** project. I suggest we don't put to much effort into this (openwifimap).

We don't use a hidden essid. I'm not sure what the owm script pushs to the db but I guess it does not push the password. I don't see a security or privacy concern in publishing interface information like name and ip address.

Could you go in to detail about why you are surprised and why it's not good practice?

sarumpaet, 2015-01-31:
I created a pull request:
https://github.com/freifunk-berlin/firmware-packages/pull/39

closed 2015-02-01 by sarumpaet