freifunk-berlin/firmware #208: redirect to wizard

cholin, 2015-01-28:
https://github.com/freifunk-berlin/firmware/commit/8dc3728a1752f96153bf254952c18357d0fbf42f tried to fix the _redirect on first boot_ bug. Problem was here that luci drops privileges during dispatching and therefor views can't use privileged functions like `ipkg.installed()` or `sys.user.checkpasswd()`.

Unfortunately this is still buggy because now there is no check if root has still no password set. Therefor the _redirect on first boot_-patch tries to automatically login by appending `?luci_username=root&luci_password=` to the url. This results in the following error message after the login-form:
```
bad argument #1 to 'getsp' (supported values: <username>) stack traceback:
 	[C]: in function 'getsp' 
 	?: in function 'getpasswd'
 	?: in function <?:791>
 	?: in function <?:147>
 	?: in function 'dispatch'
 	?: in function <?:194>
```

Reason for that is that in the login form itself `luci_username` and `luci_password` are set twice.
Once in the form via input elements and second in the REQUEST_URI (see 
https://github.com/openwrt/luci/blob/41d2b33087da393453c45f5d923d690c88ee5474/modules/luci-base/luasrc/view/sysauth.htm#L9). Luci tries to combine both (GET and POST) in a dict but during authentication it expects a string (not a dict) and therefor it crashes.

A solution would be to filter `luci_username=root&luci_password=` in `REQUEST_URI` in _modules/luci-base/luasrc/view/sysauth.htm_. This should have no side effects because at that moment automatic login procedures already failed.

booo, 2015-01-28:
The redirect is only broken if you enter a password before you try a redirect.

sarumpaet, 2015-05-26:
What about introducing an ffwizard.passwordSet uci parameter additionally to ffwizard.runBefore, set "passwordSet" to true at the very start of the Assistent, and do not redirect if passwordSet? This should cover most cases (e.g., the user setting the password but rebooting the router before finishing the setup)? Or how did you end up with password set but runBefore==false?

commit 0e240ed8 (2016-11-03): feeds: update packages-berlin

- fix #208: redirect to wizard
  https://github.com/freifunk-berlin/firmware/issues/208
- fix #381: too high tx-power on NSM2 (maybe others too)
  https://github.com/freifunk-berlin/firmware/issues/381
closed 2016-11-02 by sarumpaet