freifunk-berlin/firmware #212: "Sharing internet" without VPN

mirko, 2015-02-03:
tl;dr: Don't enforce aN VPN when deciding to 'Share internet'.

I will try to haul off as little as possible: I'm pretty sure not allowing to just share your plain internet connection without VPN was more a political rather than a technical decision.
However the situation is changing: there are statements in favour of us, also the 'ISP privilege' got interpreted in favour of us.
Although the VPN was/is a nice workaround, it's not the solution. And it's not Freifunk (if it doesn't have to be). I'd like to encourage people to 'expose' themselves, since the risk of getting sued gets smaller and smaller. Hiding is not necessary anymore.
I'd herewith with suggest adding an option for sharing the internet without needing an VPN. Although displaying a huge warning about potential consequences when selecting.

andrenarchy, 2015-02-03:
+1 for making the VPN optional!

booo, 2015-02-03:
Related to https://github.com/freifunk-berlin/firmware/issues/144

melle, 2015-02-24:
Would it be sufficient to revert patches/011-luci-freifunk-policyrouting-berlin.patch in case I want to brew my own binary without vpn?

cholin, 2015-02-24:
The easiest way is to add your default route to the `olsr-tunnel` routing table. You may use @booo's [rudimentary hotplug script](https://github.com/freifunk-berlin/firmware/issues/144) for that.

sarumpaet, 2015-10-07:
Duplicate of #144

SvenRoederer, 2017-10-29:
see also https://github.com/freifunk-berlin/firmware/issues/455

SvenRoederer, 2017-10-29:
I outlined a solution in https://github.com/freifunk-berlin/firmware-packages/pull/125

SvenRoederer, 2017-12-25:
this will be implemented by https://github.com/freifunk-berlin/firmware/pull/498 and https://github.com/freifunk-berlin/firmware-packages/pull/125

And it's tested to work by independent testers

SvenRoederer, 2018-01-30:
merged in 8664d2c after testing of at least 2 moths

closed 2018-01-30 by SvenRoederer