freifunk-berlin/firmware #213: Assistent (Internet teilen)

sebkur, 2015-02-03:
Habe eben meinen ersten Knoten selbst per Wizzard eingerichtet, dabei ist mir folgendes Usability-Problem aufgefallen:
Wenn man auf "Internet teilen" geht, kommt die Maske, in der man (a) vpn-cert und vpn-key hochläd und (b) die einstellungen zum Teilen vornimmt.

Mir war irgendwie als Benutzer nicht klar, dass ich bei (b) was angeben muss. Ich dachte mir, wenn ich bei "Benutzerdefinierte Einstellungen" nichts ankreuze, dann muss ich da auch nix angeben. Ich denke suggeriert wurde das vor allem, weil die darunter gelegenen drop-down-boxen grau sind? Wirkte auf mich so, als wären die deaktiviert und würden erst aktiv, wenn man die Checkbox ankreuzt.

Die Fehlermeldung, die bei "Weiter" kommt war jedenfalls leider auch nicht hilfreich, da die bloß stur attestiert, ein benötigtes Feld sei leer, nicht aber welches...
Vielleicht wäre es praktisch, den User darauf hinzuweisen, welches Feld auszufüllen ist (rote Umrandung oder so?)

Vielleicht besteht das Problem ja auch gar nicht mehr, habe mit 0.0.0 getestet...

sarumpaet, 2015-04-07:
Das kommt vom luci-bootstrap-Theme und ist in der ganzen Oberfläche so. Für select-Inputs wird dort das CSS von "uneditable-input" wiederbenutzt (luci-theme-bootstrap/htdocs/luci-static/bootstrap/cascade.css um Zeile 480), entsprechend wird der Text ausgegraut. Ich find's auch wenig gelungen an der Stelle.

sarumpaet, 2015-04-07:
Das hier an assistent.css anfügen wäre ein schneller Hack (der dann immerhin für den Assistenten gilt):

```css
select {
  color: #303030;
  background-color: #f8f8f8; 
}
 
input {   
  color: #303030;
}
```

closed 2015-04-29 by sarumpaet