freifunk-berlin/firmware #214: Flashspeicher einsparen

stargieg, 2015-02-08:
Hi
Packete die besonders gross und ersetzbar oder nicht notwendig sind für geräte mit wenig Flash.

### ppp 108579 Byte
Wird das Gerät als DSL Router eingesetzt oder ist der WAN Port dhcpclient?

### dnsmasq 92775 Byte
Kann durch odhcpd ersetzt werden.

### tc 91675 Byte
Ist Trafficshaping wirklich notwendig und Sinvoll? 
- Router Durchsatz
- Router CPU Leistung
- VPN Durchsatz

### ip 68752 Byte
Kann durch netifd ersetzt werden.

### iptables 49661 Byte
Kann durch fw3 ersetzt werden.

Die Byte werte stammen von einem Build für target=atheros openwrt-atheros-whr-hp-ag108
Das rootfs darf hier nicht grösser als 2228224 Byte sein.

cholin, 2015-02-08:
Danke für die Auflistung! Auf den 4MB targets ist `ppp` bereits nicht mehr enthalten (siehe [packages/vpn_4mb.txt](https://github.com/freifunk-berlin/firmware/blob/master/packages/vpn_4MB.txt#L7)). Soweit ich weiß kann man `odhcpd` keine host Dateien übergeben (bei uns die Daten aus dem olsr nameservice plugin). Dadurch würde unser mesh dns nicht mehr funktionieren. `tc` ist sicher nen Diskussion für sich und `ip` ist zum debuggen schon sehr nett. OpenWRT sollte lieber mal ihre `ifconfig` Abhängigkeiten mit `ip` ersetzen ;)

Hast du mit `fw3` schon in OpenWRT rumgespielt und Erfahrungswerte?

stargieg, 2015-02-08:
fw3 ist das standart firewall packet in openwrt.
This package provides a config-compatible C implementation of the UCI firewall.

Das du IP zum Debugen brauchst ist kein grund es in die Firmware einzubauen.
Ifconfig wird AFAIK schon jetzt durch netifd ersetzt.

Für dns sollte/gibt es auch einen openwrt ersatz. Ich weis ihn aber gerade nicht.
Ich denke aber das kadnode der richtige ersatzt für freifunk ist.
kadnode 40083 Byte

cholin, 2015-02-24:
Vorgeschlagene Ideen werden in #203 mit einfließen.

closed 2015-02-24 by cholin