freifunk-berlin/firmware #215: kmod-usb-storage & kmod-fs-ext4 Packete fuer usb storage

gruelfin, 2015-02-09:
Hi,
es wäre wirklich supertoll wenn jemand die Packete kmod-usb-storage & kmod-fs-ext4 bauen koennte, denn dann könnte ich einen usb speicher als externes medium verwenden.
Wenn ich die Packete manuell installieren will kommt folgender Fehler:

 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-usb-storage:
 *      kernel (= 3.10.49-1-0114c71ed85677c9c1e4911437af4743) *         kernel (= 3.10.49-1-0114c71ed85677c9c1e4911437af4743) * 
 * opkg_install_cmd: Cannot install package kmod-usb-storage.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-fs-ext4:
 *      kernel (= 3.10.49-1-0114c71ed85677c9c1e4911437af4743) * 
 * opkg_install_cmd: Cannot install package kmod-fs-ext4.

Danke!

stargieg, 2015-02-09:
Gute Idee.
Hier meine Packetwunschliste:
```
kmod-nls-utf8
kmod-nls-iso8859-1
kmod-nls-iso8859-15
kmod-usb-storage
block-mount
usbutils
zlib
kmod-usb-acm
usb-modeswitch
usbreset
dosfsck
fdisk
e2fsprogs
btrfs-progs
kmod-fs-btrfs
kmod-fs-cifs
kmod-fs-ext4
kmod-fs-hfsplus
kmod-fs-isofs
kmod-fs-vfat
kmod-fs-xfs
samba36-server
luci-app-samba
```

cholin, 2015-02-10:
Ich wäre dafür *Select all packages by default* (`CONFIG_ALL=y`) für `stable` builds zu setzen. Damit sollten alle Packete gebaut werden (siehe auch #93).

booo, 2015-02-11:
Ich finde das unnötig, weil wir uns dann mit allen Packages auseinandersetzen müssen, die nicht gebaut werden können. Wenn es aber allgemein gewünscht ist, wäre es super, wenn wir ein Ticket für den Buildbot eröffnen. Vielleicht können wir auch mal überlegen welche Optionen wir im Makefile abbilden können, z.B. CONFIG_ALL=y

cholin, 2015-02-12:
Warum müssen wir uns dann mit den Packeten auseinandersetzen? An sich sollte eine `make IGNORE_ERRORS=m` die einfach ignorieren (siehe [Wiki](http://wiki.openwrt.org/doc/howto/obtain.firmware.generate#building_the_image_generator)).

booo, 2015-09-07:
@Weil wir nicht auf der einen Seite sagen können, ja wir bauen die Paket und auf der anderen Seite, es ist uns egal, wenn Sie nicht bauen (es einen Fehler im build gibt).

Ich würde jetzt erstmal @sarumpaet Vorschlag mergen und das Ticket damit schließen.

closed 2015-09-12 by booo