freifunk-berlin/firmware #216: WLAN crashes TL WR1043ND V1

ces92, 2015-02-11:
The WLAN crashes often, the best way is a cronjob that restart the router every 24 h. This only a problem of the TL WR1043ND.

booo, 2015-02-11:
What do you mean by "the WLAN crashes often"? We need a more detailed description of the problem to deal with the problem.

cholin, 2015-02-12:
I think this is related to #182.

sarumpaet, 2015-02-14:
#217 might also be related if monitoring has been turned on. Another possible cause might be some IRQ related problem that got mentioned on the [mailinglist](http://lists.berlin.freifunk.net/pipermail/berlin/2015-February/027053.html); allegedly OpenWrt [r43974](http://git.openwrt.org/?p=openwrt.git;a=commitdiff;h=a3e485a8788da7e3c50ed22b98e6094dcf48e54b) fixes the problem. Unfortunately, that fix seems not to be in 14.07 yet.

ces92, 2015-02-15:
The description from the mailinglist is correct. There is also a solution on the [mailinglist] (http://lists.berlin.freifunk.net/pipermail/berlin/2015-February/027242.html), with 0.0.1 build 484(?) it disappears. I will test this build and give feedback.

sarumpaet, 2015-02-21:
Testing [build 523](http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx/523/default/) which contains the r43974 kernel patch would be highly appreciated. Have TFTP ready in case it doesn't work though.

ces92, 2015-02-26:
with build 523 it works well

sarumpaet, 2015-04-07:
@ces92 - still happy with the kernel fix in build 523? If yes we should probably include it in master.

ces92, 2015-04-14:
it's better than before but still crashes after some days

sarumpaet, 2015-05-02:
The fix is now in master and will be in the next release. If the problem persists, please reopen, with more details (logs, etc.).

closed 2015-05-02 by sarumpaet