freifunk-berlin/firmware #217: Memory leak in collectd/monitoring

sarumpaet, 2015-02-14:
In some setups, collectd sprouts a memory leak, leading to the router rebooting every few hours (for a 32MB RAM router; probably 128MB will last much longer). Turning off the "network" plugin or disabling monitoring completely solves the issue. Restarting collectd every 30mins is also fine. The issue seems to be related to kernel/file buffers or similar; neither "ps" nor /proc/PID/status show increased memory usage by collectd, but killing and restarting it does free a lot of memory. The strange thing is that I've seen routers not affected by this issue suddenly getting affected by it when just moving to another network setup; might be related to packet loss or IPvX problems? I'm investigating further.
[image]

andrenarchy, 2015-02-14:
I can confirm this issue on several routers. Example: [derrida](http://monitor.berlin.freifunk.net/detail.php?p=memory&c=&pi=&t=memory&tsi=&h=derrida&s=2678400&x=800&y=350)

cholin, 2015-02-16:
Can you confirm the cause like the collectd assumption or may it be related to #192 and #192?

andrenarchy, 2015-02-16:
Yes, collectd is causing the trouble.

cholin, 2015-02-16:
We could test [v5.4.1](https://github.com/openwrt/packages/commit/1b0bd008387d929f536cd6330bc3ee3168244e32) of `collectd` or add a cronjob to restart `luci_statistics` every 30 minutes as a hotfix.

booo, 2015-02-16:
The current version of collectd is old stable. I suggest we move to stable aka v5.4.1 and give it a try.

andrenarchy, 2015-02-21:
The current workaround is broken, see https://github.com/freifunk-berlin/firmware-packages/commit/d8e5fa79c7c5413318876475dd8b097952c50ed6#diff-aff94c28ad0798ab8304ff8595377925R59

sarumpaet, 2015-02-21:
@cholin collectd 5.4.1-1 from openwrt git does *not* fix the issue unfortunately. I guess we can close #222 then. I'll file an upstream bug report and suggest to keep/fix the restart workaround. @andrenarchy , can you commit the fix?

sarumpaet, 2015-02-23:
https://github.com/freifunk-berlin/firmware-packages/pull/47

cholin, 2015-02-24:
Did someone test #222 or how do you know this? I'm fine with freifunk-berlin/firmware-packages#47.

andrenarchy, 2015-02-24:
I have to correct my [comment above](https://github.com/freifunk-berlin/firmware/issues/217#issuecomment-74380009): on [derrida](http://monitor.berlin.freifunk.net/host.php?h=derrida) (v0.0.0) it seems *not* to be related to collectd. Restarting collectd in a cronjob doesn't fix the memory exhaustion. However, the symptoms are exactly as @sarumpaet described it: there's no process eating the memory.

cholin, 2015-02-24:
But you know that collectd saves all statistics locally in tmpfs (see `/tmp/rrd/`)?!? Higher memory consumption therefor is fine. If you don't want this, then disable `luci_statistics.rrd_tool`

cholin, 2015-02-24:
Btw if we update the luci feed we need #222 anyway because luci_statistics in master of luci depends on newer versions of collectd.

sarumpaet, 2015-02-24:
It's probably two separate issues then. I don't see growing /tmp here; disabling RRD doesn't fix the memleak for me; the 5.4.1 I installed from OpenWrt snapshots doesn't fix the issue for me.

andrenarchy, 2015-02-25:
Disk usage on `/tmp/rrd` is constant, so that's not the problem.

@sarumpaet: does restarting collectd really fix the issue for you?

sarumpaet, 2015-02-26:
@andrenarchy Yes; attached you see memory on a machine that restarts/reloads collectd every hour. It'll crash and reboot about every two hours without the cronjob. I'm seeing that behaviour on two 842s and worse on a 1043.
[image]

closed 2015-04-06 by sarumpaet

sarumpaet, 2015-04-06:
It turned out that this is not actually a memory leak per se: It's just that the 32MB RAM routers are low on RAM (they restart already if used RAM exceeds 23MB), and collectd is merely the last straw then. Interestingly, the collectd_ping plugin has RAM ramp-up behaviour (its RAM usage will increase for about two hours after start but then stay constant). The "memory leak" seen in the graph above is the start of that; the router will restart before ramping up finishes. In the fix, ping and local rrdtool (which occupies some hundred KB in the /tmp ramdisk as well) get disabled for 32MB RAM routers.

The reason for the problem being influenced by differences in network setups (see initial report) was that OpenVPN was enabled in some setups only, which causes differences in RAM load.