freifunk-berlin/firmware #218: olsrd port für collectd stimmt nicht

SvenSFS, 2015-02-14:
Beim Aktivieren von Collectd und Olsrd stimmen die Ports 
vom Plugin olsrd_txtinfo nicht überein.

In der Standard Konfiguration von OLSRD wird der Port 2007 genommen.
Aber Collectd verwendet den Port 2006.

Erst ein manuelles anpassen lässt beide zusammen laufen.
Gefunden in Kathleen 0.1.0

sarumpaet, 2015-02-22:
That seems to be a bug in https://github.com/freifunk-berlin/firmware-packages/blob/master/utils/freifunk-berlin-olsrd-defaults/uci-defaults/freifunk-berlin-olsrd-defaults . @cholin I don't know how the setup is intended to work, especially with one olsrd and one olsrd6 process. Probably we also should configure txtinfo to only accept connections from localhost.

cholin, 2015-02-24:
There was no intention to use the olsrd plugin for `luci_statistics` at all because the generated statistics seemed to be useless. But if we support this we should change the ports in `luci_statistics` and not in `olsrd`.

andrenarchy, 2015-02-25:
Why do you think the olsrd stats seem to be useless?

cholin, 2015-02-27:
Without these collectd patches from @stargieg on http://monitor.berlini.freifunk.net `LinkQuality` statistics were nearly not readable. But ok now you may can derive informations...

booo, 2015-04-30:
I guess this is fixed by now.

sarumpaet, 2015-05-01:
https://github.com/freifunk-berlin/firmware-packages/pull/48 hasn't been merged yet.

closed 2015-05-15 by booo