freifunk-berlin/firmware #219: Same SSID for 2,4 Ghz and 5 Ghz is problematic for mac clients

l0rn, 2015-02-15:
The SSID "berlin.freifunk.net" is configured on both wireless devices (2,4/5Ghz) after going through the wizard. Mac clients with dual band wireless adapters seem to have problems with that and react with frequent disconnects. So maybe it would be a good idea to alter the 2,4 band ssid to berlin.freifunk.net-legacy or something like that

cholin, 2015-02-28:
Can someone confirm this? If it's really the case I would vote to rename the ssids.

booo, 2015-03-01:
I can confirm that some clients e.g. apple products have problems with same SSID in 2 and 5 Ghz.

lynxis, 2015-03-13:
@booo please examine this problem more deeper. If we got a problem with apple device this is a serious problem. However I don't believe the problem reside in the dual ssid functionality.

booo, 2015-03-23:
@lynxis any ideas what to look for? Do we have a mac for testing somewhere?

sarumpaet, 2015-03-25:
https://support.apple.com/en-lb/HT202068 "On dual-band Wi–Fi routers, configure both bands to have the same settings"
http://superuser.com/questions/362366/does-it-make-sense-to-keep-different-ssids-for-2-4ghz-and-5ghz-wireless-networks
These links don't seem to indicate general problems with the same SSID on both bands.

What are the problems exactly? Are you sure the problem isn't related to your device seeing two *different* routers (and therefore roaming instead of re-connecting including getting new DHCP settings)?

sarumpaet, 2015-04-07:
No comment? Closing this then. @l0rn please reopen if necessary.

closed 2015-04-07 by sarumpaet