freifunk-berlin/firmware #226: OpenVPN MTU issues with KD (and possibly other providers)

sarumpaet, 2015-02-20:
The default MTU size for OpenVPN doesn't work with some providers/modem configurations, see http://lists.berlin.freifunk.net/pipermail/berlin/2015-February/027280.html and other postings. OpenVPN provides the "mtu-test" option that is supposed to be a flexible fix for this, at the cost of some additional set-up time. Opinions?

andrenarchy, 2015-02-20:
Maybe these users can test THW Option first?

shony666, 2015-02-20:
just tell me how
KD ipV6 nonbridged default KD router here and 1043 with VPN for FF

lynxis, 2015-02-23:
The mtu discovery works. But testing is even better.
@andrenarchy do we have a testing openvpn server for this?

basros, 2015-03-21:
A new vpn03 server (vpn03c) has been installed a few months ago. If vpn03-admins don't have time to configure it as additional productive server, it could be used for testing purposes. 
Ask @booo for access

booo, 2015-03-23:
Has somebody tested the mtu-test/discovery option? @lynxis ?

sarumpaet, 2015-06-29:
mtu-test unfortunately really only tests (and outputs the result to the log but doesn't change anything).

sarumpaet, 2015-07-02:
MTU set to 1300 by default in https://github.com/freifunk-berlin/firmware-packages/commit/1820c73c815a7da50e7ddd4931d4497d7c2ebd79 / kathleen 0.1.2.

closed 2015-07-02 by sarumpaet