freifunk-berlin/firmware #227: Geräteunterstützung für TP - CPE210

snakekick, 2015-02-23:
Hallo, leider unterstützt die Berliner Freifunk Firmware das TP  - CPE210 noch nicht.
Die Bielefelder Jungs sind da schon etwas weiter, Gerät wird da unterstützt. 
http://www.freifunk-bielefeld.de/?page_id=123

Eine Anpassung ist dann evtl nicht so kompliziert? Stelle mich gerne auch als beta Tester bereit.

cholin, 2015-02-23:
Freifunk Bielefeld hat seine Patches von Freifunk Gluon. Die haben haben einfach folgende zwei Patches gebackported:

* [ar71xx: add support for TP-LINK CPE210/220/510/520](http://git.openwrt.org/?p=openwrt.git;a=commit;h=3df7464f5369896e02797865bf9fdb20d0e652ac)
* [firmware-utils: add new tool tplink-safeloader for the new TP-LINK Pharos devices](http://git.openwrt.org/?p=openwrt.git;a=commit;h=6d132edd4ca10b29e9fe9272ef7ab145a289bacf)

siehe dazu [freifunk-gluon/gluon:gluon/patches/openwrt](https://github.com/freifunk-gluon/gluon/tree/master/patches/openwrt). An sich kein Problem bloß bis jmd dazu die Zeit findet ist unklar.

sarumpaet, 2015-02-23:
Vielleicht sollten wir einfach alle Patches von Gluon übernehmen...? Die Arbeit muss man sich ja nicht zweimal machen, und es sind auch andere neuere Router bei den Gluon-Patches dabei.

cholin, 2015-02-23:
Nicht alle Patches machen für alle Communities sind. Noch dazu müssen die Patches eh angepasst werden, da die Pfade nicht stimmen. Ich würde abraten blind Sachen ungetestet zu übernehmen. Oft sind die Patches auch von der zugrundliegenden OpenWRT Version abhängig. Gluon basiert auf `79465bfe7633bb92e7ea3468e9cec8bf5d4c0190` während Berlin auf `7d01d821b05b26304ab8f8fc148e690cb5ffd8b3` aufbaut. An sich sollten aber die Patches für TP-Link CPE-Geräte kein Problem sein.

Längerfristig sollte sich sicherlich der Gedanke gemacht werden, inwiefern so eine Arbeit nicht redudant gemacht werden muss. Zum Beispiel könnte Freifunk Berlin auch Gluon als Basis nehmen. Denke so ein Thema ist aber eher was für das WCW im Mai.

sarumpaet, 2015-02-23:
WCW klingt gut. - Was die Geräte angeht, bestimmt kommt bald jemand mit einem TP-Link Archer, dann geht das wieder von vorne los. ;-)

snakekick, 2015-02-23:
also mir würde der 210er erstmal reichen ;)

snakekick, 2015-02-24:
ok, kann ich also davon ausgehen das da in nächster Zeit nichts draus wird und ich das Gerät lieber umtauschen sollte?

cholin, 2015-02-24:
Die Frage ist was du unter *nächster Zeit* verstehst. Bei Freifunk geht das alles ein wenig langsamer und initiativbasierender. Muss halt jmd finden der sich dem annimmt. Sind gerade halt eh schon viele Issues und für die Firmware selbst fühlen sich halt nur 3-4 Menschen sporadisch zuständig :smile: 
Denke aber in den nächsten 2-3 Wochen sollte der Patch schon integriert sein.

cholin, 2015-02-28:
Mit f6f69de7b44d83226d7d69a32516fca5b0ac9ddc sollten die TP-LINKS CPE210/220/510/520 unterstützt werden. Siehe dazu http://buildbot.berlin.freifunk.net/builders/ar71xx/builds/532

Wenn der Build erfolgreich durchläuft, dann sollte ein testbare Firmware unter folgendem Link zu finden sein: http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx/532/

Ich hab kein Gerät zum testen, wäre also schön wenn das jmd anders übernehmen könnte.

snakekick, 2015-03-07:
hallo, sorry für die späte Rückmeldung,
die Firmware scheint zu funktionieren, bin jedenfalls gerade über das
gerät/freifunk/vpn drin.
gibt es dinge die ich darüber hinaus testen soll?

closed 2015-03-09 by andrenarchy

booo, 2015-03-13:
Ich kann auch bestätigen, dass das Image erfolgreich auf einen CPE510 geflasht werden kann.

commit f6f69de7 (2015-02-28): Backport support for TP-LINK CPE210/220/510/520

For this backport the following commits had to be backported from OpenWRT trunk
(chaos chalmer):

* 3df7464 "ar71xx: add support for TP-LINK CPE210/220/510/520" [1]
* 6d132ed "firmware-utils: add new tool tplink-safeloader for the new TP-LINK
           Pharos devices (CPE210/220/510/520)" [2]

Fixes #227.

[1] http://git.openwrt.org/?p=openwrt.git;a=commit;h=3df7464f5369896e02797865bf9fdb20d0e652ac
[2] http://git.openwrt.org/?p=openwrt.git;a=commit;h=6d132edd4ca10b29e9fe9272ef7ab145a289bacf
PR #237 (2015-03-09): Features/backport cpe
Backport support for TP-LINK CPE210/220/510/520 …
For this backport the following commits had to be backported from OpenWRT trunk
(chaos chalmer):

* 3df7464 "ar71xx: add support for TP-LINK CPE210/220/510/520" [1]
* 6d132ed "firmware-utils: add new tool tplink-safeloader for the new TP-LINK
           Pharos devices (CPE210/220/510/520)" [2]

Fixes #227.

[1] http://git.openwrt.org/?p=openwrt.git;a=commit;h=3df7464f5369896e02797865bf9fdb20d0e652ac
[2] http://git.openwrt.org/?p=openwrt.git;a=commit;h=6d132edd4ca10b29e9fe9272ef7ab145a289bacf