freifunk-berlin/firmware #228: [firmware] we don't check returncode of the imagebuilder

lynxis, 2015-02-23:
If one profile (e.g. ubnt) fails, the imagebuilder doesn't fail in all cases.

lynxis, 2015-02-24:
It's a little bit strange. On my local machine firmware-berlin fails, but on buildbot doesn't.
Any ideas?

cholin, 2015-02-24:
Maybe `-jX` related? Did you try it without jobs parameter?

lynxis, 2015-02-25:
@cholin can you try out the branch lynxis/fix-unreported-imagebuilder-fail ?

lynxis, 2015-02-25:
-jX doesn't change anything on my build host. I'm building by hand.

cholin, 2015-02-25:
Doesn't work here either. Besides lots of `Ignoring feed 'routing' - index missing` I get `Makefile:1227: recipe for target 'modules' failed`. For a full ouput see [here](http://paste.debian.net/hidden/95443aa2/).

lynxis, 2015-02-25:
Try a complete clean - `make clean` as well as `cd openwrt; make clean`. This failure isn't related to my changes.

cholin, 2015-02-25:
The failure was due an uncommited old `feeds.conf`. I checked out the original one. No routing errors anymore but it fails anyway with the same error message. If you look at the linked posted you see I built with `make clean && make -j5 V=s`. See http://paste.debian.net/hidden/dc1dea2a/.

cholin, 2015-02-25:
Master builds fine and the only difference is your https://github.com/freifunk-berlin/firmware/commit/28be96633f0c55436c4484930ef2406723571f0d commit. When I add this line to the Makefile your branch builds as well.

lynxis, 2015-02-25:
Ok. Fine. But this branch should fail because of missing dependencies in firmware creation (openvpn)

lynxis, 2015-02-26:
It works as expected. 28da92bab4 (branch fix-unreported-imagebuilder-fail_test) fails on the buildbot.
There is another bug in the imagebuilder: The IB doesn't fail when some dependencies can not be resolved.
But it will fail on missing packages, when they are selected by the package.txt

commit dd345082 (2015-02-23): [Makefile] fix fail of imagebuilder doesn't fail the build

In some cases the imagebuilder can fail on some profiles but
doesn't report back the fail to make.
Fix #228.
commit eb44ed15 (2015-02-23): [Makefile] fix fail of imagebuilder doesn't fail the build

In some cases the imagebuilder can fail on some profiles but
doesn't report back the fail to make. Fix #228.
PR #233 (2015-03-09): [Makefile] fix fail of imagebuilder doesn't fail the build
In some cases the imagebuilder can fail on some profiles but
doesn't report back the fail to make. Fix #228.
closed 2015-03-09 by andrenarchy