freifunk-berlin/firmware #232: Remove freifunk-policyrouting

andrenarchy, 2015-02-26:
Do we still need the freifunk-policyrouting package? Isn't a static configuration via http://wiki.openwrt.org/doc/uci/network#ip_rules enough for our purposes?

booo, 2015-02-26:
There is a bug in the netifd ip rule implementation. Ip rules don't work for wireless interfaces.

Apart from the bug the package is quite useful to enable/disable the functionality. With plain ip rules it's harder to configure the system behaviour.

andrenarchy, 2015-02-26:
@booo: can you add a link to the upstream bug report (if there is one)?

cholin, 2015-02-28:
I don't like the behaviour of policyrouting: changing config files. I think we do need a package for setting these rules but it could be a mich simpler one or better integrated with and without vpn setups.

booo, 2015-03-01:
@andrenarchy I reported the bug to Felix and to the mailing list of OpenWrt some time ago. I can try to create a real bug report in the next days.

Maybe we should rewrite the plugin to use only a hotplug script? This would remove the change of configuration files and reduce complexity because logic is only in one place.

andrenarchy, 2015-03-01:
@cholin: full ack, changing configuration files in a daemon/service a is a no-go.

@booo: a bug report would be helpful. Maybe we can help to fix the bug upstream.

FreifunkUFO, 2016-05-19:
i tried to deactivate the policy-routing inside berlin-firmware some weeks ago.
its crazy and doesnt work in a simple way, because the daemon/service doesnt change the configuration file (olsr) anymore :-(