freifunk-berlin/firmware #236: Remove firewall and qos packages from backbone

andrenarchy, 2015-03-08:
Usually the backbone sites don't need the firewall or QOS packages so I vote for removing them. It seems that the firewall is pulled in via `luci-mod-freifunk`.

cholin, 2015-03-08:
See as well https://github.com/freifunk-berlin/firmware/issues/52#issuecomment-75763624 for additional informations about `luci-mod-freifunk`.

basros, 2015-05-30:
On `0.2.0-alpha`, firewall is activated.  Since this most often breaks a clean upgrade path, firewall should be disabled by default on backbone images.

basros, 2015-09-23:
Is firewall and QoS really a dependency? 
If so, can't we just disable the firewall, or just remove the rules?

sarumpaet, 2015-09-23:
luci-mod-freifunk pulls in freifunk-firewall (but that isn't a dependency really). I can patch that out, but we should fork/break up luci-mod-freifunk into several packages eventually.
qos-scripts isn't a dependency of anything (it is included in default packages as it's part of its packages list) and isn't included in backbone images (anymore) as far as I can see.

booo, 2015-09-24:
@sarumpaet Can you give the break up of luci-mod-freifunk I try? The last time I tried I failed miserably.

sarumpaet, 2015-09-26:
@booo Yes. I'll create a luci-mod-freifunk-ui package that will basically be luci-mod-freifunk minus the updater and JSON functionality. I'll try to break it into -ui-public and -ui-admin but not sure if that's easily possible.

sarumpaet, 2016-04-25:
Should be fixed by https://github.com/freifunk-berlin/firmware-packages/commit/482bf96930d60af327d29c6f102c69815ce92842 etc

closed 2016-04-25 by sarumpaet