freifunk-berlin/firmware #238: sysupgrade auf kathleen 0.1.0 verliert OpenVPN bbbvpn-Zertifikate

wyae, 2015-03-10:
Das Update auf einem TP-Link TP-1043ND v1 per sysupgrade-File von http://buildbot.berlin.freifunk.net/buildbot/stable/0.1.0/ar71xx/default/ über Weboberfläche tut zwar, aber mit kleineren Tücken:

bbbvpn verliert seine Zertifikate (alle: ca, key, crt)

Nach deren Einspielen und manuellem OpenVPN-starten (über Weboberfläche) und tut's dann wieder - aber sind erst mal weg, und damit auch die OLSR-Anbindung.

booo, 2015-03-10:
Wo lagen die Zertifikate und der Schlüssel?

wyae, 2015-03-10:
Da das Einklickern über die Weboberfläche nicht funktionierte und auch die Anleitung nach http://bbb-vpn.berlin.freifunk.net/cgi-bin-index.html zu keinem Erfolg führte, habe ich das schließlich nach
http://cholin.spline.de/freifunk/bbb_vpn.html aufbauen können.

Die Zertiufikate lagen entsprechend in /etc/openvpn/

cholin, 2015-03-10:
bbbvpn ist Handarbeit und wird von der Firmware bzw dem Wizard nicht out-of-the-box unterstützt. Zur Zeit werden folgende Zertifkate bei einem Sysupgrade gesichert:

```
/etc/openvpn/freifunk_client.crt
/etc/openvpn/freifunk_client.key
```

Siehe dazu [luci-app-ffwizard-berlin/lib-upgrade/openvpn](https://github.com/freifunk-berlin/firmware-packages/blob/73306cab7b477078554fc96a4399973ca815add8/utils/luci-app-ffwizard-berlin/lib-upgrade/openvpn) und [luci-app-ffwizard-berlin/Makefile#L53](https://github.com/freifunk-berlin/firmware-packages/blob/73306cab7b477078554fc96a4399973ca815add8/utils/luci-app-ffwizard-berlin/Makefile#L53). Wir könnten natürlich auch mittels glob/wildcards `*.crt` und `*.key` einfach alle Zertifkate und Schlüssel sichern.

andrenarchy, 2015-03-10:
+1 für wildcards! Vllt auch noch `*.conf`?

wyae, 2015-03-10:
seconded.
Dazu dann auch noch die *.ca damit die Zertifikatsketten erhalten bleiben.

Danke!

Volker

booo, 2015-03-10:
Spricht etwas dagegen den ganzen Ordner `/etc/openvpn` zu behalten?

andrenarchy, 2015-03-10:
@booo: nein. +1 für kompletten `/etc/openvpn`.

cholin, 2015-03-10:
In `/etc/openvpn` liegt auch das `ffvpn-up.sh`-Skript. Das sollte nicht beibehalten werden bei einem sysupgrade. Würde mich als gegen den ganzen Ordner aussprechen.

@andrenarchy  wieso brauchst du `*.conf`, wenn diese doch aus der UCI-Konfiguration generiert werden? 
@wyae  von welchem ca sprichst du? CA und Cert von BBB-VPN und VPN03 haben jeweils eine `.crt`-Dateiendung (*freifunk-ca.crt*, *freifunk_client.key*,*bbb-vpn-ca.crt*,*bbb-vpn_X.crt*) und würden mit den Wildcards schon gesichert werden.

andrenarchy, 2015-03-10:
@cholin: es gibt in UCI auch die Möglichkeit eine config à la
```
option config /etc/openvpn/my-vpn.conf
```
anzugeben.

cholin, 2015-03-10:
Dann migrier deine Konfiguration nach UCI. Wir sollten nicht den Anspruch haben jegliche Konfigurationsparameter zu unterstützen.

wyae, 2015-03-10:
Ah! Sorry, käsiges Gedächtnis: die CA-Zertifikate heißen ja üblicherweise *-ca.crt und nicht *.ca - sind also schon in der *.crt-Wildcard mit 'drin.

lynxis, 2015-03-10:
Wieso ist denn /etc/openvpn/ nicht automatisch im sysupgrade includiert? Fehlt ein bb backport?Im trunk ist das schon so:
https://github.com/openwrt-mirror/openwrt/blob/master/package/network/services/openvpn/files/openvpn.upgrade

booo, 2015-03-10:
@lynxis wird von uns überschrieben: https://github.com/freifunk-berlin/firmware-packages/blob/73306cab7b477078554fc96a4399973ca815add8/utils/luci-app-ffwizard-berlin/lib-upgrade/openvpn

lynxis, 2015-03-10:
@boo dann weg weg mit unserem ueberschreiben. Der default ist gut!

booo, 2015-03-23:
Ich denke auch, dass die Upgrade-Konfiguration von openwrt für openvpn gut ist, aber wir wollen wirklich nicht das `ffvpn-up.sh` backupen. Macht es Sinn, wenn wir das nach /lib/functions/ verschieben? Gibt es einen besseren Ort?

lynxis, 2015-05-03:
/usr/lib/freifunk/ oder /lib/freifunk/ würde ich benutzen.

wyae, 2015-07-10:
Das Problem besteht auch beim Update auf Kathleen 0.1.2

booo, 2015-07-19:
@wyae Danke fürs Testen. Im nächsten Release sollte das Problem gelöst sein.

closed 2015-07-21 by sarumpaet