freifunk-berlin/firmware #239: Disable public AP if no internet connectivity

sarumpaet, 2015-03-11:
If the node does not have internet connectivity, the public AP ("berlin.freifunk.net") should get disabled.
Mailinglist discussion: http://lists.berlin.freifunk.net/pipermail/berlin/2015-February/027058.html
Weimar's implementation: https://github.com/bittorf/kalua/blob/aed9167c5f4abf9552ecda65695a99da37ea5c21/openwrt-addons/etc/kalua/net#L1194

booo, 2015-03-11:
Even though internet access is one of our core services a node without internet is still of value for the user. I don't think we should disable the AP if the internet connection is gone!

sarumpaet, 2015-03-11:
Just in case - I'm not talking about the ad-hoc interface, I'm talking about "berlin.freifunk.net", and changing its name in case of no internet connectivity (as done in the Weimar firmware) is one option.

Thing is that people associating to "berlin.freifunk.net" should always get working internet - but currently that's quite often not the case (see mailinglist discussion). This gives bad PR.

booo, 2015-07-19:
I still think it is not a good idea to disable or change the ap ssid of the node if internet is not available. Freifunk should work even without internet. It's confusing if the ssid changes from time to time. It complicates debugging. And we create yet another script that can cause bugs.

If you don't mind I will close this ticket.

sarumpaet, 2015-07-21:
I guess we have to discuss what "berlin.freifunk.net" means at some point. It surely doesn't mean "BBB access" or "ICVPN access" anyways since "island" meshes (connected via VPN) do not provide that, and there are quite a lot of these. If "berlin.freifunk.net" doesn't mean "Internet access" either, little remains.

I second that such functionality needs to be robust and transparent (and optional, be that opt-in or opt-out). It might tie in better with some proper logging framework; atm debugging an unreliable connection isn't much fun as the firmware doesn't provide easily readable end user logs.

But anyways, this functionality won't implement itself, and if nobody else joins this discussion, please feel free to close the ticket. :)

booo, 2015-09-12:
I'm closing the ticket as you suggested.

I suggest you create a ticket for your logging idea so we can work on it.

closed 2015-09-12 by booo