freifunk-berlin/firmware #240: [Statistiken: Schnittstellen] Gerätebezeichnungen fehlen :: Versuch 2

yjebos, 2015-03-16:
In den Statistiken zu den Schnittstellen fehlen die Bezeichnungen.
So ist leider nicht zu erkennen, für welches Gerät die Kurve erstellt wurde.
Bsp: "Transfer on"

__Firmware:__ Freifunk Berlin kathleen 0.1.1
__Hardware:__ TP-Link TL-WR1043N/ND v2 
__Welche Statistiken? (LuCI? Monitor? URL?)__
    Im LuCI: Statistiken -> Diagramme -> Schnittstellen
    Über jeder Kurve steht "(hostname): Transfer on" bzw. "(hostname): Packets on" allerdings ohne die Schnittstelle selbst zu benennen.
**Für (hostname) steht der richtige Hostname drin**

In der __Firmware:__  *OpenWrt Attitude Adjustment pberg-221*  werden diese Info angezeigt. Bsp: "(hostname): Packets on wlan0-1"

basros, 2015-03-21:
Ein ähnliches Verhalten bzgl. fehlender Interface-Namen habe ich mit Ubuntu 14.04 collectd-nodes auf monitor.berlin.freifunk.net festgestellt:
Anscheinend werden die Interface-Stats bei einer neuen Version von collectd nicht wie früher unter 
```<node-name>/interface/if_octets_<iface>.rrd``` abgelegt, sondern unter  ```<node-name>/iterface-<iface>/if_octets.rrd```

sarumpaet, 2015-05-08:
Ja, das ist wohl auch ein Relikt vom neueren LuCI, das von collectd 5 ausgeht, wir aber noch collectd 4 benutzen. Siehe https://github.com/freifunk-berlin/firmware/issues/253 . Ich schließe das hier mal.

closed 2015-05-08 by sarumpaet