freifunk-berlin/firmware #241: owm.lua Fehlermeldung NanoM2 Kathleen 0.1.0

berndnemetz, 2015-03-17:
Hallo, 

folgende Pakete sind installiert:

luci-app-owm - 0.4.10 - Luci JSON Export for Open Wireless Map
luci-app-owm-cmd - 0.4.10 - luci-app-owm-cmd - Commandline update tool
luci-app-owm-gui - 0.4.10 - luci-app-owm-gui - GUI Open Wireless Map

wenn ich owm.lua ausführe bekomme ich folgende Fehlermeldung:

root@ap6-0-bernd:~# owm.lua 
/usr/bin/lua: ?:0: attempt to index field 'wirelessdevice' (a nil value)
stack traceback:
	[C]: in function 'foreach'
	?: in function 'get'
	/usr/sbin/owm.lua:65: in main chunk
	[C]: ?

Das Setup verfügt neben dem AdHoc Interface auch über einen virtuellen AP im Managed Modus und einer Kabelkopplung. Auf einer anderen Nano gleiche FW Version ABER OHNE Kabelkopplung gibt es keine Probleme.

carsifreifunk, 2015-03-18:
Gleicher Fehler kommt auch beim WDR4300, mit Kathleen 0.1.0, ohne Kabelkopplung zustande. 

Trotz der Fehlermeldung wird der Node korrekt auf der OWM-Karte angezeigt.

lynxis, 2015-04-22:
@berndnemetz 
kannst du bitte /etc/config/network und /etc/config/wireless (ohne wlan pws) anhängen?

berndnemetz, 2015-04-22:
Hallo

anbei die gewünschten Daten. Passt das so? 

NETWORK:

config interface 'loopback'
	option ifname 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fdea:00a2:475e::/48'

config interface 'wan'
	option ifname 'eth1'
	option proto 'dhcp'
	option peerdns '0'
	option delegate '0'

config interface 'wan6'
	option ifname '@wan'
	option proto 'dhcpv6'
	option peerdns '0'

config rule 'olsr_allif_ipv4'
	option lookup 'olsr'
	option priority '1000'

config rule 'localnets_allif_ipv4'
	option lookup 'localnets'
	option priority '2000'

config rule 'olsr_tunnel_tunl0_ipv4'
	option lookup 'olsr-tunnel'
	option priority '19999'
	option in 'tunl0'

config rule 'olsr_default_tunl0_ipv4'
	option lookup 'olsr-default'
	option priority '20000'
	option in 'tunl0'

config rule 'olsr_default_unreachable_tunl0_ipv4'
	option action 'unreachable'
	option priority '20001'
	option in 'tunl0'

config interface 'ffvpn'
	option ifname 'ffvpn'
	option proto 'none'
	option delegate '0'

config interface 'dhcp'
	option proto 'static'
	option type 'bridge'
	option _orig_ifname 'eth0 wlan0-dhcp-2'
	option _orig_bridge 'true'
	option ipaddr '6.6.0.1'
	option netmask '255.255.255.0'
	option delegate '0'
	option dns '213.73.91.35 8.8.8.8'

config rule 'olsr_tunnel_wireless0_ipv4'
	option lookup 'olsr-tunnel'
	option priority '19999'
	option in 'wireless0'

config rule 'olsr_default_wireless0_ipv4'
	option lookup 'olsr-default'
	option priority '20000'
	option in 'wireless0'

config rule 'olsr_default_unreachable_wireless0_ipv4'
	option action 'unreachable'
	option priority '20001'
	option in 'wireless0'

config rule 'olsr_tunnel_dhcp_ipv4'
	option lookup 'olsr-tunnel'
	option priority '19999'
	option in 'dhcp'

config rule 'olsr_default_dhcp_ipv4'
	option lookup 'olsr-default'
	option priority '20000'
	option in 'dhcp'

config rule 'olsr_default_unreachable_dhcp_ipv4'
	option action 'unreachable'
	option priority '20001'
	option in 'dhcp'

config rule 'olsr_tunnel_ffvpn_ipv4'
	option lookup 'olsr-tunnel'
	option priority '19999'
	option in 'ffvpn'

config rule 'olsr_default_ffvpn_ipv4'
	option lookup 'olsr-default'
	option priority '20000'
	option in 'ffvpn'

config rule 'olsr_default_unreachable_ffvpn_ipv4'
	option action 'unreachable'
	option priority '20001'
	option in 'ffvpn'

config interface 'wireless0'
	option proto 'static'
	option ipaddr '10.22.6.0'
	option netmask '255.255.0.0'
	option dns '85.214.20.141 213.73.91.35 8.8.8.8'
	option delegate '0'

config interface 'LANKOP'
	option proto 'static'
	option ifname 'eth0'
	option ipaddr '10.22.255.5'
	option netmask '255.255.255.252'
	option delegate '0'

config rule 'olsr_tunnel_LANKOP_ipv4'
	option lookup 'olsr-tunnel'
	option priority '19999'
	option in 'LANKOP'

config rule 'olsr_default_LANKOP_ipv4'
	option lookup 'olsr-default'
	option priority '20000'
	option in 'LANKOP'

config rule 'olsr_default_unreachable_LANKOP_ipv4'
	option action 'unreachable'
	option priority '20001'
	option in 'LANKOP'

WIRELESS:

config wifi-device 'radio0'
	option type 'mac80211'
	option path 'pci0000:00/0000:00:00.0'
	option htmode 'HT20'
	option disabled '0'
	option country 'DE'
	option diversity '1'
	option hwmode '11g'
	option chanbw '20'
	option txantenna 'all'
	option rxantenna 'all'
	option channel '5'
	option txpower '10'

config wifi-iface
	option ifname 'wlan0-adhoc-2'
	option network 'wireless0'
	option mode 'adhoc'

config wifi-iface
	option ifname 'wlan0-adhoc-2'
	option network 'wireless0'
	option device 'radio0'
	option mode 'adhoc'
	option ssid 'www.freifunk-potsdam.de'
	option bssid '02:ca:ff:ee:ba:be'
	option encryption 'none'

config wifi-iface
	option ifname 'wlan0-dhcp-2'
	option encryption 'none'
	option device 'radio0'
	option mode 'ap'
	option ssid 'Freifunk-Potsdam-6-0'
	option network 'dhcp'

lynxis, 2015-04-23:
danke. ich hab mal kurz in owm reingeschaut. ich brauch doch noch mehr files. kannst du mir /etc/config/* per mail an  lynxis@fe80.eu schicken? github supported leider nur bilder als attachment.
dann kann ich einfach deine files direkt auf eine nano m2 rueberkopieren und testen.

berndnemetz, 2015-04-23:
> danke. ich hab mal kurz in owm reingeschaut. ich brauch doch noch mehr files. kannst du mir /etc/config/* per mail an lynxis@fe80.eu <mailto:lynxis@fe80.eu> schicken? github supported leider nur bilder als attachment.

Hallo Alexander,

anbei die gepackte Konfig.

Wenn Du noch was brauchst, schreist Du einfach :-)

Gruß Bernd

btw: Danke für Deine/Eure Arbeit. Leider kann ich an dieser Stelle nicht helfen, weil keine Ahnung :-/

lynxis, 2015-04-24:
Hi Bernd,
du musst die Mail direkt an mich schicken. Github wirft Anhänge weg.
Gruß
lynxis

booo, 2015-09-12:
Ich schlage vor, dass wir das Ticket schließen, wenn es keinen weiteren Fortschritt mehr gibt.

berndnemetz, 2015-09-17:
Hi,

Von meiner Seite keine Einwände, nach einem flashen auf Kathleen 0.1.2 läuft alles.

---
Gruß Bernd

sokai, 2016-04-08:
Hey … – Gerade habe ich den Bug mit einer (frischen) ```kathleen 0.1.2``` (Reboot hilft auch nicht) erneut und stehe etwas auf dem Schlauch, da hier keine Lösung zu finden ist … :-/

Der manuelle Aufruf von ```/usr/sbin/owm.lua``` erzeugt folgenden Output:
```bash
/usr/bin/lua: ?:0: attempt to index field 'wirelessdevice' (a nil value)
stack traceback:
	[C]: in function 'foreach'
	?: in function 'get'
	/usr/sbin/owm.lua:65: in main chunk
	[C]: ?
```

Und wenn ich http://frei.funk/cgi-bin/luci/owm.json aufrufe, erhalte ich das:
```bash
/usr/lib/lua/luci/dispatcher.lua:448: Failed to execute call dispatcher target for entry '/owm.json'.
The called action terminated with an exception:
?:0: attempt to index field 'wirelessdevice' (a nil value)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:448: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:195: in function </usr/lib/lua/luci/dispatcher.lua:194>
```
http://frei.funk/cgi-bin/luci/owm/ wird aber problemlos angezeigt, nur der Node mit diesem Problem fehlt in der Kart …

→ Gibts ne Lösung? (Gern kann ich auch nochmal Configs senden …)

Danke und Grüße!

SvenRoederer, 2017-10-03:
in https://github.com/freifunk-berlin/firmware-packages/pull/109 gab es einen Umbau am OWM-code.
Besteht das Problem bei euch noch?

SvenRoederer, 2019-12-18:
Nach mehr als 2 Jahren ohne Feedback, vermute ich dass Problem besteht nicht mehr.
Fals doch, das Issue wieder auf machen.

closed 2019-12-18 by SvenRoederer