freifunk-berlin/firmware #243: Wrong Links in Main-Menu

basros, 2015-03-22:
Version: ```0.1.0 -  0.1.1-alpha```

Some links in the upper main menu are broken:
Depending on which menu-entry has been selected, some other menu-entries will link to previous-entry + anker.
Map + Services seem to OK.
I did not manage to get to the status page at all!

booo, 2015-03-22:
I guess this is related to https://github.com/freifunk-berlin/firmware/issues/197

sarumpaet, 2015-05-07:
It took me a bit to figure out what basros actually meant. I guess the expected behaviour is to get to some Status page when clicking on the "Status" menu entry in the public router UI. I wasn't even aware such a page exists (it's at /freifunk/status); I thought the "Status" menu was merely badly named (and should be called "Diagnostics" or something, with the Test Download in it). :)

Seems the menu system doesn't support menu (titles) that both represent a page and a menu title. See the "OLSR" entry - it's merely a headline. So this is a bug in luci-mod-freifunk (upstream). It can be fixed by patching /usr/lib/lua/luci/controller/freifunk/freifunk.lua with an additional
> entry({"freifunk", "status", "status"}, template("freifunk/public_status"), _("Status"))

line. I wonder why https://github.com/openwrt/luci/blob/master/modules/luci-mod-freifunk/luasrc/controller/freifunk/freifunk.lua doesn't have this. Perhaps there's a LuCI/luci-mod-freifunk version mismatch?

basros, 2015-06-13:
Issue seems to be related to bootstrap:
> 15:31 < ufo-v4> hab grad mal theme-bootstrap installiert.. ja, da geht "status" auch nicht mehr... mit dem freifunk-theme gehts

Kudos to @FreifunkUFO for testing!

booo, 2015-07-19:
@sarumpaet can you fix this upstream?

sarumpaet, 2016-05-06:
In openwrt/feeds/luci/themes/luci-theme-bootstrap/luasrc/view/themes/bootstrap/header.htm:
```
<li class="dropdown">                                             
<a class="menu" href="#">
```
-- replace "#" by "<%=pcdata(href)%>"
See also https://github.com/openwrt/luci/issues/713 - not sure if we have some version mixup there.