freifunk-berlin/firmware #245: LuCI profile_berlin updates

sarumpaet, 2015-03-25:
https://github.com/freifunk-berlin/luci/tree/freifunk_berlin_updates collects profile_berlin updates to be pushed upstream. Not sure about some things. We should probably keep olsrd defaults in upstream? And introduce olsrd6 defaults?
@cholin @booo @andrenarchy

sarumpaet, 2015-04-06:
Updated the commit to keep OLSR and OWM settings.

booo, 2015-07-19:
@sarumpaet Why did you keep the owm settings? I removed them once because the defaults where sane for our setup.

We can keep the olsrd defaults in upstream and we should maybe introduce olsrd6 defaults. The problem here is that we don't use these defaults anyway. In the worst case they create broken configurations with other assistants. Maybe we should add a comment?

sarumpaet, 2015-07-21:
@booo OWM: Ok, it wasn't clear that the OWM package defaults to api.openwifimap.net already; as with the other settings I thought the OWM parts got removed since they were replaced by uci-defaults. Still, an indication that we use OWM (and not some other map) would be nice...

booo, 2015-07-22:
Why do we need the indication that we use the OWM? I don't think there is software e.g. meshkit that makes any use of this information. If we just need a bit of documentation for the user we should add this to the README.

sarumpaet, 2016-02-07:
Further discussion see #310

closed 2016-02-07 by sarumpaet