freifunk-berlin/firmware #253: collectd ping statistics broken

sarumpaet, 2015-05-06:
Both in monitor.berlin.freifunk.net and in the router web interface, ping statistics are broken (first graph - ping rtt I guess - does not render, the other two graphs are empty). Apparently the plugin doesn't send data (yes, I'm aware I disabled the ping plugin for 32MB RAM routers - I'm looking at that on a 3600); the rrd file typically does not get created (but rarely it *is* created).

As a side note, for routers sharing their internet connection, the ping plugin should probably use the ffvpn interface (plugin option "Device", but that's not handled properly yet by the uci wrapper /usr/bin/stat-genconfig).

sarumpaet, 2015-05-08:
With the default "Interval 1" it works. The RTT graph is missing because the collectd ping plugin outputs its values as type "ping", not "value" (even in newer collectd versions, and despite types.db not agreeing). This needs a patch for /usr/lib/lua/luci/statistics/rrdtool/definitions/ping.lua (I'll do that) and fixes on monitor.berlin.freifunk.net (@booo?).

sarumpaet, 2015-05-08:
Actually, it seems the ping plugin outputs the type "ping" (standard in collectd4), but collectd seems to rewrite the type according to the types.db, so it becomes "value" (collectd5 standard). rrd files get initialized only once though, so their type will stay at "ping" unless migrated. https://collectd.org/wiki/index.php/V4_to_v5_migration_guide#Migrating_existing_RRD_files - I guess on monitor.berlin.freifunk.net either rrd files haven't been migrated yet, or the types.db still uses the collectd4-style "ping" type, or the collectd4 network plugin doesn't honor types.db.

sarumpaet, 2015-05-15:
For some reason even our current collectd4 values end up as "value" type in the RRD files on monitor.berlin, which failed to parse them since its CONFIG[version] is set to 4 (thus trying to parse "ping" types). I patched plugin/ping.php for the time being.

sarumpaet, 2015-06-01:
Concerning the plugin not working at all:
* The 30 secs ping interval is too high. With a bit of packet loss, servers will not get any data.
* It seems the ping plugin tries to ping the IPv6 address of monitor.berlin.freifunk.net even if we do not have IPv6 connectivity to that host (e.g., via VPN03).

sarumpaet, 2015-08-03:
We should not hardwire monitor... as ping target but set up a dedicated ping.berlin.freifunk.net DNS name. This gives us a bit more flexibility. @booo , can you set this up? I suggest using 77.87.49.66 for this for the time being (or actually anything that is not on in-berlin due to its current packet loss). Once that's done, I'll write migration.

booo, 2015-08-03:
I do not have access to the dns configuration. We should ask @egmont1227 to setup up a dns entry. I can write a mail if you like.

sarumpaet, 2015-08-03:
Yes please, and thanks in advance.

booo, 2015-09-12:
@sarumpaet We now have a special ping dns for berlin:
```
➜  ~  dig +short ping.berlin.freifunk.net
77.87.49.66
```
I changed the default configuration. We still need a migration.

booo, 2015-10-26:
@sarumpaet do you want to work on the "other" problems here or can we close this one?

sarumpaet, 2015-10-27:
I'm undecided whether ping should go through wan or ffvpn (for VPN nodes). Also, last time I checked I think the "Device" collectd_ping setting didn't quite work, so closing this for now.

closed 2015-10-27 by sarumpaet

booo, 2015-10-27:
Maybe it's possible to specify two ore more ping targets/ways so we can ping both? If you think this is import let us create a new and specific ticket.