freifunk-berlin/firmware #254: set SmartGatewayDividerEtx in olsrd default configuration

booo, 2015-05-15:
We should set the SmartGatewayDividerEtx to somthing different from 0. 'If SmartGatewayDividerEtx is zero then no weighing [of the gateway] is performed' [1]. We can set the value in the olsrd default configuration package:

https://github.com/freifunk-berlin/firmware-packages/blob/master/utils/freifunk-berlin-olsrd-defaults/uci-defaults/freifunk-berlin-olsrd-defaults#L55

[1] https://github.com/servalproject/olsr/blob/master/README-Olsr-Extensions#L271

sarumpaet, 2015-05-23:
https://github.com/freifunk-berlin/firmware-packages/tree/smartgw-divideretx contains that and migration.
However, I'm not sure the 4096 default recommended in http://www.olsr.org/git/?p=olsrd.git;a=blob;f=README-Olsr-Extensions;h=b5ce50fe45078cfdf6891db4d6f7cb7e01867992;hb=master#l376 is a good choice for us; it seems to put a high emphasis on gateway bandwidth, mostly ignoring path cost, and it's recommended for the etx_ffeth metric (but we use etx_ff). Opinions?

booo, 2015-07-22:
I suggest we try 4096. Maybe we can try it on a production node in the network?

booo, 2016-05-08:
I suggest that we try to get this into the new release.

lynxis, 2016-08-06:
**mostly ignoring path cost**?!
I suggest dropping all non-criticial patches on olsrd and put more efforts in olsr2