freifunk-berlin/firmware #256: Package OLSRv2

basros, 2015-05-23:
After WCW2015, we are more eager then ever to push new mesh technologies!

Please package `OLSRv2` (aka OONF) using `git://olsr.org/oonf.git`

If you encounter complications during build-process because of plugin `hash_tomcrypt` depending on `libtomcrypt`, then just don't include this plugin.

sarumpaet, 2015-06-02:
It's in the chaos-calmer branch/build 623. http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx/623/
MPC85XX didn't build but this seems to be a buildslave issue or possibly new dependencies.

closed 2015-06-02 by sarumpaet

sarumpaet, 2015-06-03:
http://buildbot.berlin.freifunk.net/buildbot/unstable/mpc85xx/550/
The buildslave had run out of disk space.