freifunk-berlin/firmware #257: Traffic limiter limits the private AP as well as ffvpn

sarumpaet, 2015-05-26:
...since it's limiting the wan interface. It should probably limit the ffvpn interface (and get triggered on ffvpn up).

booo, 2015-07-19:
The limiter should only limit the ffvpn interface. I guess we can change the interface here to ffvpn: https://github.com/freifunk-berlin/firmware-packages/blob/master/utils/luci-app-ffwizard-berlin/luasrc/tools/freifunk/assistent/ffwizard.lua#L44

booo, 2015-09-12:
Why should we trigger the qos on ifup events?

sarumpaet, 2015-09-30:
The WAN interface might be down on boot or disappear temporarily (connected modem crashed, etc.). I just saw that qos-scripts uses an ifup script so fortunately this already gets handled. I'll update the defaults and write migration.

closed 2015-10-26 by booo