freifunk-berlin/firmware #263: olsr6 policy routing broken with BBB

milchprodukt, 2015-07-15:
default firmware 0.1.2 is not able to route public ipv6 with BBB, since there is no ploicyrouting set up but instead olsrd6 is configured for dedicated tables.

You need to comment the following lines in /etc/config/olsrd6 to get ipv6 working in BBB

```
       option RtTable '111'
       option RtTableDefault '112'
       option RtTablePriority '1000'
       option RtTableDefaultOlsrPriority '20000'
```

lynxis, 2016-08-06:
IMHO: I like to kick off every olsr1 ipv6 setups. but yes it's broken.

lynxis, 2016-08-09:
But if somebody sent patches, I'm fine with those.

sarumpaet, 2016-11-03:
Anyone got more comments on this? @basros ? Anything against just commenting out the lines indicated in the first comment? @booo ?