freifunk-berlin/firmware #272: use ifup/down events to start openvpn

booo, 2015-09-01:
At the moment openvpn starts on system boot. It starts even if we do not have a connection on the WAN port (to the internet). In the worst case openvpn traffic gets routed through the mesh (through other openvpn tunnels). In one of three cases this can happen because the firewall configuration for openvpn/vpn03 misses one of the server ips [1].

We should start openvpn on wan ifup events and make sure there is a default route via the wan interface. If the interface goes down we should stop openvpn. Once we have this script we can remove the old firewall rules.

[1] https://github.com/freifunk-berlin/firmware-packages/blob/master/utils/freifunk-berlin-openvpn-files/uci-defaults/freifunk-berlin-openvpn#L38

sarumpaet, 2015-09-30:
+1!
As a side note, the OpenVPN "local <ip>" directive allows binding OpenVPN to specific IPs/interfaces, which would be another failsafe in the /etc/hotplug.d/iface/openvpn script.

sarumpaet, 2015-10-05:
https://github.com/freifunk-berlin/firmware-packages/commits/openvpn-hotplug
Needs a bit of testing.
(without "local" for the time being)

booo, 2015-10-05:
I will do some testing and report back. Thanks for the work!

sarumpaet, 2015-10-05:
I added some logging and handling of the "local" parameter. This should make sure OpenVPN only connects through WAN in all cases. I guess we can assume the WAN interface always has an IPv4 address? Is there a better way to get its address? Removing "nobind" has no side effects? ("local" conflicts with it - on first glance I think leaving away nobind makes debugging easier, too)

I added some logging. The idea is to put some "[ff]" style logs in other scripts and make logread|grep "[ff]" (or a persistent version of that) visible in the web interface eventually. It should also help making the setup self-documenting.

sarumpaet, 2015-10-06:
@booo - concerning the inline comments (commenting here since that's easier to follow)
* I guess we can do without "local", but then in some edge cases OpenVPN might still connect via mesh, e.g., on ifup WAN but it doesn't have a default route; possibly on ifup WAN but nonworking connection (DSL down, etc.).
* I don't like changing config in scripts either, but I don't see a better way here. I'm tempted to ignore the UCI OpenVPN stuff altogether and just write the OpenVPN start command directly into the ifup script: There's nothing we gain from all the OpenVPN UCI config rewrite stuff, but we lose easy maintainability and transparency (everybody with basic Unix knowledge can change the OpenVPN start call; probably nobody can edit the OpenVPN UCI config without looking at UCI docs and OpenVPN UCI shell scripts additionally) plus have problems with dynamic options (i.e., "local"). In the ifup script, we can do something like "if uci.openvpn.enabled then openvpn start else openvpn --local $WANIP --mssfix 1300..." so that the user can still override the behaviour. Bonus points: We can easily add examples for alternate configs to the script (commented out), e.g., TCP config.
* Logging prefix: Yes, as I explained above, the idea is to use the [ff] prefix from now on for user relevant logging *eventually*. However, changing other places (e.g., migration) is another PR.

sarumpaet, 2015-10-07:
I'll implement the "if uci.openvpn.enabled then openvpn start else openvpn --local $WANIP --mssfix 1300..." thing. This allows us to use --local cleanly, is easy to maintain, and cuts out all the migration and UCI config maintenance annoyances (these may make sense for other things - if users customize settings - but not for FF-OpenVPN, which is static).

sarumpaet, 2015-10-07:
I'll also try to merge #144 .

sarumpaet, 2015-10-09:
@cholin Was there anything wrong with https://github.com/freifunk-berlin/firmware-packages/commit/1a7c71ff3e132ed77ddc271d7b625b8acfde8b59 ? Otherwise the openvpn-hotplug branch should not require iptables anymore. :)

cholin, 2015-10-09:
If it's still in there it should be working otherwise ipip tunnels for SmartGateway clients would not work.

sarumpaet, 2015-10-09:
I just cherry-picked that commit; it never made it to master. Ok, I'll just test it (and write migration).

cholin, 2015-10-09:
:smile:

sarumpaet, 2015-10-09:
Hm. There's pretty much the same (iptables) stuff in /lib/functions/olsrd.sh (which is called on olsrd start, see "olsrd: /etc/init.d/olsrd: olsrd_setup_smartgw_rules() Notice: Inserting firewall rules for SmartGateway" log message). I wonder why we need that firewall.user line.

Also, just adding "tnl+" to the freifunk zone doesn't seem to work; iptables-save|grep tnl is empty, and SGW doesn't work. "fw3 network eth0" complains with "Warning: Section 'zone_freifunk' cannot resolve device of network 'tnl+'" - I guess it's not that easy...

cholin, 2015-10-09:
You're right.The firewall.user line is outdated since 2001 (https://github.com/openwrt-routing/packages/commit/059b26c242ad5a97f7c51b9fd68784073abb2a53). We do not need to alter firewall rules anymore (https://github.com/openwrt-routing/packages/blob/master/olsrd/files/olsrd.init#L722).

sarumpaet, 2015-10-11:
The tunnel stops working if I delete the line in firewall.user. I guess this is because wireless ifup events may occur after olsrd was started, but these events trigger a firewall reload which flush the changes the olsrd init script made...?

Also, fortunately fw3 has a syntax for non-UCI devices: http://wiki.openwrt.org/doc/uci/firewall#zone_declaration_for_non-uci_interfaces - adding
> option device 'tnl_+'

to the freifunk zone does the job. :)

SvenRoederer, 2016-04-18:
I have something build on this in my unstable-builds for some time (24, March 2016) and not heard of any user complains yet.
The only point I found is that OpenVPN binds to port 1194 per default. Put a quick hack in place to use a random port to bind to.

sarumpaet, 2016-04-25:
fixed by https://github.com/freifunk-berlin/firmware-packages/commit/59791e2594215bfdb3306daa9dbe3685059bd30a etc

closed 2016-04-25 by sarumpaet