freifunk-berlin/firmware #274: change dyn gw ping ip

booo, 2015-09-01:
We should change the dynamic gateway ping ip. It's a random ip that I don't recognize:

https://github.com/freifunk-berlin/firmware-packages/blob/master/utils/freifunk-berlin-olsrd-defaults/uci-defaults/freifunk-berlin-olsrd-defaults#L43

If this ip is unreachable all gateways will stop to announce gateway functionality in the mesh.

booo, 2015-09-03:
https://github.com/freifunk-berlin/firmware-packages/commit/f0713de1812db4a4149f4a4001b85f71d4cb0c90#commitcomment-13051795

booo, 2015-09-05:
Maybe it should be an ip address in our control or an ip address that never goes down like 8.8.8.8.

sarumpaet, 2016-04-13:
According to http://olsr.org/git/?p=olsrd.git;a=blob;f=lib/dyn_gw/README_DYN_GW dyn_gw supports multiple IPs (in priority order), so I guess we should add some FF Berlin specific IPs first and only end with 8.8.8.8.

booo, 2016-04-13:
I'm in favor of dns root servers: https://de.wikipedia.org/wiki/Root-Nameserver

SvenRoederer, 2016-04-18:
What about setting up a DNS-name with multiple records for the servers to use and having some script changing the config and DNS-lookup per boot or WAN-hotplug?

sarumpaet, 2016-04-20:
We don't need to make that overly complicated. What about berlin.freifunk.net (or some other FF Berlin server that has a reliable IP), VPN servers, our configured DNS servers, root DNS servers, 8.8.8.8, in that order?

sarumpaet, 2016-04-25:
fixed in https://github.com/freifunk-berlin/firmware-packages/commit/17b1dd1652595fbf148cc658dcc2157f4fd39b3e

closed 2016-04-25 by sarumpaet