freifunk-berlin/firmware #275: Update der Community Profile (openwrt/luci)

FreifunkUFO, 2015-09-07:
Bitte aktualisiert mal die Community Profile in openwrt/luci (derzeit sind die nur als Patch vorhanden)
Sind diese noch meshwizard/meshkit - kompatibel?

Lt. https://github.com/freifunk-berlin/firmware/blob/master/patches/013-delete_non_existing_profiles.patch können die Profile "Pberg und Berlin NordOst" gelöscht werden!?

FreifunkUFO, 2015-09-07:
siehe auch https://github.com/freifunk-berlin/firmware/issues/245

booo, 2015-09-12:
Eins der Profile ist jetzt verschwunden. Das Pberg-Profile kann eigentlich auch weg, oder @stargieg ?

Kompatibilität zum meshwizard/meshkit testen wir zur Zeit nicht und werden wir in Zukunft auch eher nicht testen. Wir empfehlen auch niemandem einen anderen Wizard als den Berliner zu nutzen.

sarumpaet, 2015-09-30:
Duplikat von #245

closed 2015-09-30 by sarumpaet