freifunk-berlin/firmware #276: Add "well-behaving" captive portal / splash screen

slomo, 2015-09-16:
We as Freifunk should be interested in informing the user about our project and call neighbours to action (and maybe also to consent to certain rules).  This might also be useful in the upcoming German wireless network legislation effort.

slomo, 2015-09-16:
I am acting more as a proxy here, the issue has been raised at the CBase meeting. I also discussed it a few weeks with @faust2k and juergen.

pr0vieh, 2015-09-20:
:thumbsup:

sarumpaet, 2015-09-26:
Is this a generic list, or are these things all broken with luci-app-splash?

slomo, 2015-09-28:
This was a very generic list. I am sure that the former Freifunk splash screen didn't obey them. We might be able to configure luci-app-splash right and get the same result.

But my intention was more to discuss whether and how (optional or not) it should be integrated. Afterwards I could also try to implement it.

lynxis, 2015-09-28:
Splashs are buggy, because OS are buggy even with some advancetages taken in the last year(s). They still forgot sometimes to detect the splash and don't inform the user, that he must act somehow. Some OS open the splash page automatically, some not. And Splash pages can only work on http:// connections, not on https://.

sarumpaet, 2015-09-28:
I'd leave it up to the node administrator to enable splash (opt-in).

@lynxis , I don't see why "OS are buggy" if they don't expect/check for IP spoofing (which splashes do after all).

I don't think a splash must work in all cases. It should provide information on a best-effort basis and must not disturb services - for the Freifunk use case. Of course, AP providers that want the splash for "legal reasons" have different requirements, but I don't see why we should bother with that.

So working only on http is fine I think, and we might even whitelist specific IPs/URLs in case specific apps have problems with the splash.

lynxis, 2015-09-28:
the main question is *how can we communicate with the user other than a splahpage*.
Splashpages are pain, because some OS don't recognize them and switch fully to this new connection and all applications fails. this is our problem when using a splashpage.

sarumpaet, 2015-09-28:
I'd be all for a better berlin.freifunk.net main page as well. :)

Are there still apps that use http instead of https? Also, note that we can whitelist these.

But it is a pain, yes.

booo, 2016-01-01:
Is someone willed to work on this and can summarize the current state of discussion and the way forward? If not I suggest that we close this ticket in the next week.

pmelange, 2018-11-04:
After over 2.5 years of inactivity, I am closing this ticket.

closed 2018-11-04 by pmelange