freifunk-berlin/firmware #278: Allow USB-Tethering

hubx, 2015-09-16:
Hallo lieber Berlin,

vielen Dank erstmal für die tolle Firmware, macht uns in Potsdam, die Arbeit an vielen Stellen leichter.

Wir machen gerade WLAN für die neue Potsdamer Erstaufnahmestelle. Nun sind beim Uplink da leider auf UMTS/LTE per Android-Tablet mit Tethering angewiesen.

Auf dem WDR3600 läuft ein kathleen 0.1.0. 
Für USB-Tethering (ähnlich #215, #93) fehlen leider folgende [Packete](http://wiki.openwrt.org/doc/howto/usb.tethering):

`kmod-usb-net kmod-usb-net-rndis kmod-usb-net-cdc-ether`
Das deckt sich auch mit dem wie sich das Tablet auf unseren Linux-kisten meldet.

Wir haben versucht Packete von https://downloads.openwrt.org/attitude_adjustment/12.09/ar71xx/generic/packages/ und https://downloads.openwrt.org/barrier_breaker/14.07/ar71xx/generic/packages/base/ nachzuladen, aber wenn wir das richtig verstanden haben basiert kathleen 0.1.0 auf `r44162`und damit konnten wir packete nicht nutzen :(

Könnt ihr uns da kurzfristig bescheid geben, ob ihr Packete für 0.1.0 nachbauen könnt? Wenn nein, auch nicht schlimm seht es bitte als Feature request für ein zukünfiges release. Wir basteln die USB-Ethernetbrücke dann mit einem anderen vanilla OpenWRT oder Linux Gerät

Danke & Grüße

hubx, 2015-09-16:
Ah, das bauen war mit Anleitung ja echt machbar 

allerdings kommt bei mir, wenn ich freifunk-berlin/firmware@0.1.0
baue kommt 
kernel_3.10.49-1-2ae811afe5887c8cbac7963de44523c5_ar71xx.ipk
anstatt
http://buildbot.berlin.freifunk.net/buildbot/stable/0.1.0/ar71xx/packages/base/kernel_3.10.49-1-c9116102f84f95eb2bb8871ae599f6b2_ar71xx.ipk
raus, damit werde ich die entstandenen kernel packete ja nicht installieren können :(

Da komme ich wohl ohne Kernel Upgrade nicht drum herum, oder? Geht das so inplace oder ist das nicht zu empfehlen?

booo, 2015-09-16:
Ich füge die Packete in den nächsten Stunden hinzau. Die landen dann in unserem neusten Release.

Du kannst wahrscheinlich die Packete installieren. Der Kernel ist im Prinzip der gleiche. Einfach mal ausprobieren (--force).

hubx, 2015-09-17:
Hi, ich konnte die compilierten kernel module reinkopieren (am package management vorbei)
Von mir aus kann das zu :)

booo, 2015-09-17:
Wir können die Packages als Module bauen, wenn du willst. Sollte kein großer Mehraufwand sein.

hubx, 2015-09-17:
Also für mich nicht mehr nötig, aber vielleicht jemand anderes :)

PR #280 (2015-09-17): build kernel modules for usb-ethernet tethering fixes #278
this allowed me to use install the modules `kmod-usb-net kmod-usb-net-rndis kmod-usb-net-cdc-ether`
and fixed #278 for me
commit 1dbb5c82 (2015-09-17): Merge pull request #280 from hubx/usb-ethernet

build kernel modules for usb-ethernet tethering fixes #278
closed 2015-09-17 by booo