freifunk-berlin/firmware #284: Upgrade 0.2.0 to recent OpenWRT Chaos Calmer 15.05

basros, 2015-09-26:
Chaos Calmer 15.05 has been officially released!

Please also upgrade OLSR (0.9.0.3), OONF (0.9.2), batman-adv (2015.1) and alfred (2015.1) to recent versions from openwrt [routing-feed].

If somehow possible, please also include [tunneldigger] or ask @lynxis for support.

[routing-feed]: https://github.com/openwrt-routing/packages
[tunneldigger]: https://github.com/wlanslovenija/firmware-packages-opkg/tree/master/net/tunneldigger

booo, 2015-09-26:
openwrt update is included in: https://github.com/freifunk-berlin/firmware/tree/update_chaos_calmer

Please test and report back.

basros, 2015-09-26:
Which build number / binary firmware on [buildbot] is based on that release? 

[buildbot]: http://buildbot.berlin.freifunk.net/buildbot/unstable/

booo, 2015-09-26:
What about this one: http://buildbot.berlin.freifunk.net/builders/ar71xx/builds/674

basros, 2015-09-26:
Successfully updated a TP-Link TL-WR741N/ND v4

Mesh-related packages are a little bit outdated:
```
root@elok-nordwest:~# batctl -v
batctl 2015.0 [batman-adv: 2015.0]

root@elok-nordwest:~# olsrd -v

 *** olsr.org - 0.9.0.2-git_b1674df-hash_3cbcb99142689db08347b3a3b0c0cbf6 ***
 Build date: 2015-09-21 20:09:52 on ubuntu
 http://www.olsr.org

root@elok-nordwest:~# alfred -v
alfred 2015.0

root@elok-nordwest:~# uname -a
Linux elok-nordwest 3.18.21 #1 Mon Sep 21 20:16:45 UTC 2015 mips GNU/Linux

```
But it even runs 3.18.21 - Thank you!

booo, 2015-10-26:
I update the chaos calmer version to the latest commit in the origin repository. I'm going to update the feeds in the next hours/days.

closed 2015-10-26 by booo