freifunk-berlin/firmware #286: Simplify config.berlin backend

sarumpaet, 2015-10-01:
I'm just adding the CPE210, and this is far too complicated and brittle.

I suggest this:
* replace http://config.berlin.freifunk.net/wizard/routers with a Wiki page
 * there we can also give more information about the individual routers
* link to http://config.berlin.freifunk.net/wizard/forms/... through that Wiki page
 * parameters to the Wizard should be router name ("TP-Link CPE210/220/510/520"), router id ("tp-cpe", used later for building the link back to the download - "https://wiki.../page#tp-cpe"), possibly the number of mesh IPs.
* everything else just as before.

This would remove most of default.cfg, a lot of code, can replace the berlin.freifunk.net download page, gives the user more information, lets users get involved easier (want to add hints to a router? Edit the wiki) and is much easier to maintain at the same time.

cholin, 2015-10-09:
Issues for http://config.berlin.freifunk.net should be addressed at https://github.com/freifunk-berlin/config.berlin.freifunk.net/issues.

Concerning your issue: I'm not a fan of wiki pages because they seem to be outdated really soon. But for me it's not that important so go ahead.

SvenRoederer, 2016-10-01:
it was discussed in the meetings and this approach got positive feedback.

even there is a risk of outdated pages, this applies to the current page also, if not even more to the current.
It was agreed that there might be a more big userbase to update wiki-pages.

closed 2016-11-28 by sarumpaet