freifunk-berlin/firmware #290: Update Feeds to 15.05 notation

stargieg, 2015-10-17:
-src-git packages git://github.com/openwrt/packages.git^e96cf8db9128e6f71cfe36cc0d986406f60d4432
+src-git packages https://github.com/openwrt/packages.git;for-15.05
Should I do this?

booo, 2015-10-17:
I think it's a good idea to pin the packages repo to a specific commit. If we only use a branch every build has another version of the packages repo. One of the core ideas of the firmware was reproducibility.

stargieg, 2015-10-17:
like this ?
https://github.com/stargieg/firmware/commit/918d84bd1b9025dc4e91ca700bf41863fabae0f2

booo, 2015-10-19:
Yes, this looks much better. I will merge this soon.

booo, 2015-10-26:
I going up update the feeds in the next hours/days. Stay tuned. Thanks for you contribution!

closed 2015-10-26 by booo