freifunk-berlin/firmware #291: Firmware für den 841v10

mmalte, 2015-10-17:
Hallo Berlin!

Hier ist jemand aufgeschlagen der auf der Firmware für den 841v10 ist, könnt ihr ihm weiterhelfen?

LG aus Aachen
MrMM

sarumpaet, 2015-10-31:
Support für den 841v10 gibt es erst seit dem 26.10.2015 im OpenWrt Master:
http://git.openwrt.org/?p=openwrt.git;a=commitdiff;h=75c0f8d51ffe17a64ccd31a0b38f3fa8c119d8dd
http://git.openwrt.org/?p=openwrt.git;a=commitdiff;h=1cd7e6c1fd07527d2a6a7660c85e7e90160956a9
Falls das nicht davon in den nächsten Tagen einen Commit auch in CC gibt, können wir evtl. den Patch von Gluon übernehmen:
https://github.com/freifunk-gluon/gluon/blob/master/patches/openwrt/0031-ar71xx-fix-ath79_soc_rev-value-for-QCA9531-ver.-2.patch

Benjarobbi, 2015-11-22:
Hi, hat sich jetzt etwas mehr getan? Gibt es mehr Infos?

cpietsch, 2015-12-12:
würde mich auch interessieren

SvenRoederer, 2016-01-10:
mit 2 commits am 5.12.2015 ist der V10-support nach CC geportet worden.
ich hab das mal bauen lassen :
+ alle anderen Änderungen for CC
+ update der community-profile für Berlin + Cottbus
+ support für TL-WR1043v3, DLink DIR505

http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx/745/

SvenRoederer, 2016-03-05:
da der o.g. ordner vom buildserver gelöscht wurde, hier ein aktueller: http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx/797

sarumpaet, 2016-04-25:
https://github.com/freifunk-berlin/firmware/commit/f45576c1dc21b3d46ac7179eab3d1fe2e8f52e39

closed 2016-04-25 by sarumpaet

theincogtion, 2016-08-18:
Gibt es wirklich noch keine Lösung? Ich wäre auch an v10 interessiert. Wenn man einen neuen Kauft hat man in der Regel v10

SvenRoederer, 2016-08-18:
@theincogtion  das Ticket ist geschlossen, weil unter o.g. commit der support für V10 für Kathleen 0.2.0 in der Firmware ist
http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx/1035/

theincogtion, 2016-08-18:
ok aber leider noch unstable?