freifunk-berlin/firmware #292: Lan-Port-Bug in Kathleen für TP-LINK CPE210

moenk, 2015-10-25:
Man muss in der Default-Konfiguration der 0.1.2 auf LAN0 das PoE und auf LAN1 das WAN anschließen. Das ist sehr unpraktiisch weil ja nur eine Leitung zum Router nach draußen gehen soll. Außerdem ist es gefährlich auf LAN0 sein WAN anzuschließen (so wie es in der Anleitung zum Router steht) weil man damit versehentlich sein Netz nach außen ungeschützt freigibt.

ber-red, 2015-10-26:
aufgefallen ist mir das auch. finde es ebenfalls verwirrend und gefährlich.
- ist es ein Fehler, oder genügt es, die Berlin Doku anzupassen?
- ist das in Kathleen zu fixen oder in openwrt?

Flasht man jedenfalls einen CPE210 mit Kathleen ist die Interface Zuordnung folgendermassen:

Beschriftung auf CPE210 | interface config | switch config
----|----|----
LAN0 (POE IN) | interface 'dhcp' , ifname 'eth0.1' | vlan '1' , ports '0t 5'
LAN1 | interface 'wan' , ifname 'eth0.2' | vlan '2' , ports '0t 4'

'Ursache' scheint, dass @NeoRaider eine Analogie gezogen hat, die nicht ganz passt:  http://openwrt-devel.openwrt.narkive.com/thdXnmiF/patch-v2-0-2-tp-link-cpe210-220-510-520
>**LAN and WAN have been switched**
>*While the choice of WAN and LAN roles is arbitrary as the ports are labeled as
LAN0 and LAN1, I've changed LAN to LAN0 (the PoE port) and WAN to LAN1 to
match the mappings used on Ubiquiti NanoStations.*

slomo, 2015-10-28:
The lines that requires changes seems to be this one:

https://github.com/openwrt-mirror/openwrt/blob/95f36ebcd774a8e93ad2a1331f45d1a9da4fe8ff/target/linux/ar71xx/base-files/etc/uci-defaults/02_network#L83

booo, 2016-01-01:
We should keep this for consistency reasons as it is. The configuration depends on the usecase and we can't cover all usecases. I'm also not sure which usecase is more common.

I suggest that we close this ticket soon.

sarumpaet, 2016-01-01:
@booo Can you explain what you mean by "consistency"?
The (only!) use case we focus on on berlin.freifunk.net, our documentation, and the wizard, is "Freifunk mesh AP connected to uplink router".
I'm against closing this ticket as clearly there's several people having problems with the current setup, and it's frequently mentioned on the mainlinglist. Just closing this ticket won't make the problem go away.

booo, 2016-01-01:
Consistency means: The CPE-configuration should be as similar to the nanostation configuration as possible. If we switch the ports on the CPE we also switch the configuration on the nanostation.

About the use case you are somehow right. But we could also think about this device as the typical device to connect users to the mesh. If they just want to use mesh and do not want to act as a gateway the configuration is somehow right.

Can someone please make a proper proposal for a change towards a more user friendly flashing/configuration process. Just switching the ports after flashing does not solve the problem.

sarumpaet, 2016-01-02:
Nitpicking: For me that's not *consistency*, it's just a shortcoming/lazyness of some behind-the-scenes code. The CPE and the NSM are different devices with different configs.

Handling the CPE port setup could be done on bootup, for example in the migration script, before it checking the guard. That'd be only a few lines. We could also make that configurable with a checkbox in the Assistent ("PoE port is a. WAN or b. FF-LAN, please choose"). I guess detecting the CPE is just a one-liner in lua/bash.

Documentation is basically #286. We don't want more documentation in git (i.e., config.berlin code/configuration) but use the wiki for that: There are quite a few routers which need some lines of documentation (e.g., the Wifi switch of the 3600 not working properly; newer NSMs needing a downgrade before flashing etc.), and we want users to be able to contribute.

sarumpaet, 2016-02-20:
This bugs also concerns the NSM Locos. There it's even better as they only have one Ethernet port.

For the record, one relevant mailinglist thread including manual reconfiguration:
http://news.gmane.org/find-root.php?group=gmane.org.freifunk.berlin&article=22463

booo, 2016-02-20:
@sarumpaet I think that (locos) is a problem that should be address upstream if possible. Is there a way to detect the loco case in software?

sarumpaet, 2016-02-23:
Why should upstream care? In contrast to us they don't even have "WAN port goes to internet" as the standard use case as far as I know but I may be wrong.

I don't have a normal NSM nor a Loco here. I guess the Loco can be detected as "if hardware==NSM and num_ethernet_ports==1" or something. ;-)

lynxis, 2016-08-06:
i don't see here an bug. the manual of the router is for the router firmware. not for our firmware. but adding more documentation would be nice!
Because this is so far just a discussion, I remove the milestone 0.2.0.

christoph-buente, 2016-11-01:
Hi,

i have a Nanostation 5 m flashed with "Powered by LuCI Trunk (git-14.294.77948-ecb0c2f) Freifunk Berlin kathleen 0.1.2"

I opened the berlin.freifunk.net wifi, but cannot share my internet connection. I do have the VNP3 credentials setup. But which ever port i try, it does not work. @lynxis where would that documentation live? Thx

SvenRoederer, 2016-11-01:
@christoph-buente  unter http://wiki.freifunk.net/Berlin:Firmware#Router-spezifische_Hinweise gibt's einen Abschnitt für CPE210. Da wird verlinkt auf: http://berlin.berlin.freifunk.narkive.com/9gl7g6uY/berlin-wireless-cpe210-ports-vertauschen-nur-in-config-vertauschen#post7 und Thomas schreibt, wie man die Ports umdreht.

christoph-buente, 2016-11-01:
Hallo Sven,

vielen Dank. Habe die CPE spezifischen Sachen immer ignoriert, da ich dachte das trifft auf die nanostation nicht zu. Ich habe mir auch den Post zum Port vertauschen durchgelesen. Jedoch finde ich im Luci den Punkt "Netzwerk => Switch" überhaupt nicht.

[image]

ich habe das `/etc/config/network` file editiert und die switch config reingeschrieben. Nun taucht auch der Menüpunkt auf. Allerdings bin ich irritiert, weil ich nur 4 statt 5 Ports habe. Und zweitens, dass die alle mit 10Base verbunden sein sollen, obwohl das Kabel in einem 100MBit switch steckt.

[image]

SvenRoederer, 2016-11-07:
@christoph-buente  ich merk grad, das ich falsch bin, du hast eine Nanostation (wie du schreibst), da hilft dir mein CPE-hinweis garnicht.
Evtl. hast du dir jetzt einen switch konfiguriert, den es gar nicht gibt :-/

bobster-galore, 2017-02-18:
On CPE210 we could detect the use case from wizard setting "take part in freifunk" -> POE = dhcp or
"share my internet" POE = wan-port. That would reduce the risk of misconfiguration.

bobster-galore, 2017-09-22:
What will it be like, when CPE210 V2 (single port Version) only is distributed?

SvenRoederer, 2017-09-23:
as on the NSMs loco: FF-LAN

bobster-galore, 2018-04-06:
Am I right, that currently the wizard is not aware of the machine it's running on?
Are there any machine dependent scripts working after running the wizard?
I could imagine checking /proc/cpuinfo against a list of machines which need a postwizard script. Where we could fumble with certain settings.
From ffwizard.settings.sharenet='1' we would know that someone want's to share her internet connection.
So in case of CPE210 (NSM whatever) = yes and sharenet = yes, we could change the switch settings.
How u mean?

SvenRoederer, 2018-04-07:
There are several ways to check for the current board (via cpuinfo, via rpc, probably via luci.system). Comparing this to a list of relevant boards when sharenet=1 is not as hard.
probably it's a good idea to implement the real action into a separate shell-script, which can be called by the wizard and even directly from the shell.

bobster-galore, 2018-04-07:
At openwrt is a [script](https://github.com/openwrt-mirror/openwrt/blob/95f36ebcd774a8e93ad2a1331f45d1a9da4fe8ff/target/linux/ar71xx/base-files/etc/uci-defaults/02_network#L83)  which does set certain values for interface, switch and vlans.
Dunno when it runs. There is already a setting for cpe510, probably we modify that script?
But only if it runs after the wizard, which is not likely?!
But it could be like a receipt how to do the sharenet=1 changes.

SvenRoederer, 2018-04-23:
Are you referencing the "/etc/board.d" scripts?
these are used for initial setup of the hardware and running much before our uci-defaults and even before the wizard

pmelange, 2018-11-04:
@bobster-galore, it looks like you have made a lot of progress on this issue with branch https://github.com/freifunk-berlin/firmware-packages/tree/lan-port-on-sharenet

I have added a new router, ubnt-unifiac-mesh, to the firmware (#611).  This device has only one port.  The default is that lan is on this port.  Would you be able to add support to switch this to wan if sharenet==1?

pmelange, 2019-02-03:
On the "switch" configuration page, it is also easy enough to just enter a "2" where VLAN ID "1" is, and "1" where VLAN ID "2" is.

SvenRoederer, 2019-02-03:
also affects UAPs (https://lists.berlin.freifunk.net/pipermail/berlin/2019-February/039053.html).
To change this setting in conjunction with "share-internet" makes sense here again.

SvenRoederer, 2019-12-28:
> @bobster-galore, it looks like you have made a lot of progress on this issue with branch https://github.com/freifunk-berlin/firmware-packages/tree/lan-port-on-sharenet

has been merged to master in https://github.com/freifunk-berlin/firmware-packages/commit/51fc9526526e6df4258c2a2e8c872e328ec770dc

SvenRoederer, 2020-01-25:
this is also relevant for #759 (TPlink MR3020 V3)

SvenRoederer, 2020-02-13:
For Gluon there was a related issue: https://github.com/freifunk-gluon/gluon/pull/1932. but it seems they can change the assignment in the "setup-mode".

SvenRoederer, 2020-04-13:
I had a chance to work on the "sharenet-switch.sh" script (https://github.com/freifunk-berlin/firmware-packages/blob/bdfe20564fbdedffe4f12e3bc6c3a78cd05b5aa3/addons/freifunk-berlin-configtools/root/usr/share/freifunk-berlin/sharenet-switch.sh). It hast the basic functionality for the NSM2 (so I expect also for NSM5) and allows swapping the assignment of the physical LAN-ports.

I think this can quickly extended to the TPLink CPEs v1. In general more testing is needed.

SvenRoederer, 2020-06-28:
_sharenet-switch_ should work for NSM, NSM Loco and CPe v1 so far. In #826 I mentioned that there is a way of creating a "wan" section missing.