freifunk-berlin/firmware #293: create ibss interfaces on first boot

booo, 2015-10-26:
Instead of creating the default ibss/adhoc interfaces in the wizard we should create these interfaces on first boot. In the current incarnation of the firmware the creation of the ibss interfaces does not depend on user input. I don't see a reason why this should be part of the wizard.

booo, 2016-01-21:
I created some functions to manipulate the wireless configuration:

```sh
radio_enable() {
        local radio=${1}
        uci set wireless.${radio}.disabled=0
}
radio_disable() {
        local radio=${1}
        uci set wireless.${radio}.disabled=1
}
radio_set_channel() {
        local radio=${1}
        local channel=${2}
        uci set wireless.${radio}.channel=${channel}
}
iface_set_device() {
        local wifi_iface=${1}
        local device=${2}
        uci set wireless.${wifi_iface}.device=${device}
}
iface_set_mode() {
        local wifi_iface=${1}
        local mode=${2}
        uci set wireless.${wifi_iface}.mode=${mode}
}
iface_set_ssid() {
        local wifi_iface=${1}
        local ssid=${2}
        uci set wireless.${wifi_iface}.ssid=${ssid}
}
iface_set_bssid() {
        local wifi_iface=${1}
        local bssid=${2}
        uci set wireless.${wifi_iface}.bssid=${bssid}
}
iface_set_network() {
        local wifi_iface=${1}
        local network=${2}
        uci set wireless.${wifi_iface}.network=${network}
}
iface_create() {
        local name=${1}
        local device=${2} #TODO maybe reorder arguments
        local mode=${3}
        local ssid=${4}
                
        local cfg=$(uci add wireless wifi-iface)
        uci rename wireless.${cfg}=${name}

        iface_set_device ${name} ${device}
        iface_set_mode ${name} ${mode}
        iface_set_ssid ${name} ${ssid}
}
iface_create_adhoc() {
        local name=${1}
        local device=${2}
        local ssid=${3}
        local bssid=${4}
        iface_create ${name} ${device} 'adhoc' ${ssid}
        iface_set_bssid ${name} ${bssid}
}
iface_create_ap() {
        local name=${1}
        local device=${2}
        local ssid=${3}
        iface_create ${name} ${device} 'ap' ${ssid}
}
```

sarumpaet, 2016-01-25:
All but two functions are merely aliases for something else? Why that? This obscures what's actually happening and makes the code hard to read for anyone who's new. If I were to read "iface_set_bssid" somewhere, I would expect that that function does exactly that (i.e., calls to iw, etc.), not that it merely changes UCI config.

SvenRoederer, 2019-12-29:
To transform the uci-changes of the script to real config, we can just call "uci-commit wireless; reload_config".

I also changed the title to include 802.11s-setups. 

In general I don't see a problem in enabling mesh by default initally.