freifunk-berlin/firmware #294: create a ipv6 mesh network by default

booo, 2015-10-26:
The routers should mesh by default using the ULA ipv6 prefix.

This issue depends on: https://github.com/freifunk-berlin/firmware/issues/293

optikfluffel, 2015-10-29:
[image]

stargieg, 2015-10-29:
> Zukunft <https://camo.githubusercontent.com/cb38b9e468f51fc592c4cc1094c0392488e5e790/687474703a2f2f7777772e7265616374696f6e676966732e636f6d2f722f667472652e676966>

-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

LOL

Vergangenheit

a) https://github.com/stargieg/packages_berlin/tree/master/utils/auto-ipv6-node

b) https://github.com/stargieg/luci2-ffwizard

-----BEGIN PGP SIGNATURE-----
Version: GnuPG/MacGPG2 v2
Comment: GPGTools - http://gpgtools.org

iEYEARECAAYFAlYx3ssACgkQr9m0OkMZoqS1VgCg3roTGaLoGc0bbfXgHphZdmIr
zB8AmwZnRt8oBEMzDLU3M+D+nZkYXHf4
=/fFf
-----END PGP SIGNATURE-----