freifunk-berlin/firmware #297: Faulty(?) IGMP snooping breaks ICMPv6 neighbor discovery

nils-wisiol, 2015-11-01:
I'm using a Freifunk-enabled TP-Link WDR4300 with "Freifunk Berlin kathleen 0.1.2 (git-14.294.77948-ecb0c2f)" for my private WiFi. The WiFi interface is bridged to the "wan" interface, which is connected to an IPv6-enabled Hitron router provided (read: enforced) by Vodafone/Kabel Deutschland. The Hitron usually answers ICMPv6 neighbor discovery packets with solicited router advertisements. However, using a network client on the WiFi provided by the Freifunk WDR4300, the ICMPv6 packets don't get through.

If I disable IGMP snooping [1] on the Freifunk bridge [2], everything works as expected. So my best guess is that the IGMP snooping (which is designed to minimize broad and multicast traffic on bridges) is somehow faulty and forwards the ICMPv6 packets to the wrong interface or not at all.

I'm not aware of any circumstances that are special to my setup, which is why I'm assuming the problem will arise with other IPv6-enabled wan-bridges as well.

(Just as a side note: fixing the problem with disabling IGMP snooping in `/proc` will only work until `ifup wan` is run, which is by default every morning at 6.)

I propose to either fix the IGMP snooping or disable it for now by default.

[1] https://en.wikipedia.org/wiki/IGMP_snooping
[2] `echo 0 > /sys/devices/virtual/net/br-wan/bridge/multicast_snooping`

booo, 2015-11-01:
Thanks for reporting the problem in such a detail!

To me this sounds like a problem we should also report upstream (openwrt, kernel?) and should be fixed there in the end. Can you create a minimal setup based on openwrt that reproduces the problem? I think this would help to report and fix the problem upstream.

In the meantime we could add a hotplug script to /etc/hotplug.d/iface/ that disables the IGMP on the interface. There seems to be an uci option too but I'm not sure if it works.

Maybe this issue is of interest: https://dev.openwrt.org/ticket/17954

nils-wisiol, 2015-11-01:
I don't have spare hardware that I can use for testing. Is there a way to do that in software?

booo, 2015-11-01:
Maybe you can use qemu for this: http://wiki.openwrt.org/doc/howto/qemu

nils-wisiol, 2015-11-02:
I'll check that out. Meanwhile, this might be related: https://bugzilla.redhat.com/show_bug.cgi?id=880035#c12

andrenarchy, 2015-11-08:
@nils-wisiol thx a lot for reporting! You saved me and @chmac a few hours of debugging! :+1: 

We should set 
```
option igmp_snooping 0
```
on the bridge interfaces (i.e. on the `dhcp` interface) by default, see [here](http://wiki.openwrt.org/doc/uci/network#options_valid_for_all_protocol_types).

nils-wisiol, 2015-12-06:
Setting `option igmp_snooping 0` on the `dhcp` interface didn't help for me, I had to set it on the `wan` interface (which is also a bridge). Now it's fixed for good. Thanks, @andrenarchy