freifunk-berlin/firmware #301: Problems with v0.1.2 on TL-WR842ND Ver2.3

webseba, 2015-11-02:
Hello! I'd like to get my TP-Link WR842ND Ver2.3 ready for Freifunk. After flashing at restating it with the accurate firmware provided on your website the software seems to work fine. However, after putting in my information and getting to the "safe and restart" button i get the following error: 

[image]

Also later when trying again and being asked for my admin password I sometimes got: 

[image]

What could all this be about. Any ideas? Thanks!

sarumpaet, 2015-11-03:
Did you disable JavaScript? see https://github.com/freifunk-berlin/firmware/issues/279
Are you sure IPs were entered correctly? (dots, not commas, etc.)

The second problem is
https://github.com/freifunk-berlin/firmware/issues/208

webseba, 2015-11-11:
Thank you for the help! Indeed, the problem was resolved after enabeling scrips and cookies.
Now there seems to be another problem with my vpn settings. Please take a look if this tells you anything: 

Wed Nov 11 22:46:08 2015 daemon.err openvpn(ffvpn)[2132]: TLS Error: TLS key negotiation failed to occur within 60 seconds (check your network connectivity)
Wed Nov 11 22:46:08 2015 daemon.err openvpn(ffvpn)[2132]: TLS Error: TLS handshake failed
Wed Nov 11 22:46:08 2015 daemon.notice openvpn(ffvpn)[2132]: SIGUSR1[soft,tls-error] received, process restarting
Wed Nov 11 22:46:10 2015 daemon.warn openvpn(ffvpn)[2132]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Wed Nov 11 22:46:10 2015 daemon.notice openvpn(ffvpn)[2132]: UDPv4 link local: [undef]
Wed Nov 11 22:46:10 2015 daemon.notice openvpn(ffvpn)[2132]: UDPv4 link remote: [AF_INET]78.41.116.65:1194
Wed Nov 11 22:46:49 2015 daemon.info hostapd: wlan0-dhcp-2: STA 00:26:bb:17:c6:42 IEEE 802.11: authenticated
Wed Nov 11 22:46:49 2015 daemon.info hostapd: wlan0-dhcp-2: STA 00:26:bb:17:c6:42 IEEE 802.11: associated (aid 1)
Wed Nov 11 22:46:49 2015 daemon.info hostapd: wlan0-dhcp-2: STA 00:26:bb:17:c6:42 RADIUS: starting accounting session 5643B6E6-00000000
Wed Nov 11 22:46:49 2015 daemon.info dnsmasq-dhcp[1575]: DHCPREQUEST(br-dhcp) 192.168.178.46 00:26:bb:17:c6:42 
Wed Nov 11 22:46:49 2015 daemon.info dnsmasq-dhcp[1575]: DHCPNAK(br-dhcp) 192.168.178.46 00:26:bb:17:c6:42 wrong network
Wed Nov 11 22:46:51 2015 daemon.info dnsmasq-dhcp[1575]: DHCPREQUEST(br-dhcp) 192.168.178.46 00:26:bb:17:c6:42 
Wed Nov 11 22:46:51 2015 daemon.info dnsmasq-dhcp[1575]: DHCPNAK(br-dhcp) 192.168.178.46 00:26:bb:17:c6:42 wrong network
Wed Nov 11 22:46:54 2015 daemon.info dnsmasq-dhcp[1575]: DHCPDISCOVER(br-dhcp) 00:26:bb:17:c6:42 
Wed Nov 11 22:46:54 2015 daemon.info dnsmasq-dhcp[1575]: DHCPOFFER(br-dhcp) 10.230.158.75 00:26:bb:17:c6:42 
Wed Nov 11 22:46:54 2015 daemon.info dnsmasq-dhcp[1575]: DHCPDISCOVER(br-dhcp) 00:26:bb:17:c6:42 
Wed Nov 11 22:46:54 2015 daemon.info dnsmasq-dhcp[1575]: DHCPOFFER(br-dhcp) 10.230.158.75 00:26:bb:17:c6:42 
Wed Nov 11 22:46:55 2015 daemon.info dnsmasq-dhcp[1575]: DHCPREQUEST(br-dhcp) 10.230.158.75 00:26:bb:17:c6:42 
Wed Nov 11 22:46:55 2015 daemon.info dnsmasq-dhcp[1575]: DHCPACK(br-dhcp) 10.230.158.75 00:26:bb:17:c6:42 
Wed Nov 11 22:47:10 2015 daemon.notice openvpn(ffvpn)[2132]: [UNDEF] Inactivity timeout (--ping-restart), restarting
Wed Nov 11 22:47:10 2015 daemon.notice openvpn(ffvpn)[2132]: SIGUSR1[soft,ping-restart] received, process restarting
Wed Nov 11 22:47:12 2015 daemon.warn openvpn(ffvpn)[2132]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Wed Nov 11 22:47:12 2015 daemon.notice openvpn(ffvpn)[2132]: UDPv4 link local: [undef]
Wed Nov 11 22:47:12 2015 daemon.notice openvpn(ffvpn)[2132]: UDPv4 link remote: [AF_INET]77.87.49.66:1194

I'm a bit confused of where to upload the .ovpn .key and the two .crt files that were sent to me. Only two of the are needed during the setup process...

sarumpaet, 2015-11-12:
You have to put in the .crt and the .key file that share the same base filename. Ignore the freifunk-ca.crt file. You have to wipe the router (System/Backup.../"Reset to defaults" in the admin UI) to re-upload these files as there's currently a bug that prevents replacing them.

closed 2015-11-12 by sarumpaet