freifunk-berlin/firmware #306: add hostname to neigh.sh script

booo, 2015-11-09:
The neigh.sh script does not show the hostname of the remote and locale site. The hostname of the remote site is only one nslookup away so we should included it.

bittorf, 2015-11-09:
you can even do this without lookup:
´´´shell
#!/bin/sh
while read line; do
  set -- $line
  case "$1" in
    "$ip") echo $2 ;;
  esac
done </var/run/hosts_olsr

booo, 2015-11-09:
@bittorf thanks for the hint.

Related source: https://github.com/openwrt/luci/blob/master/contrib/package/freifunk-common/files/usr/bin/neigh.sh

bittorf, 2015-11-18:
habe ein ziehbegehren erstellt:
https://github.com/openwrt/luci/pull/550

can you please show me the output of "/var/run/hosts_olsr" with IPv6?

booo, 2015-11-18:
Output should look like this:

```
root@stuttgarter-core:~# cat /tmp/hosts/olsr.ipv6 
### this /etc/hosts file is overwritten regularly by olsrd
### do not edit

127.0.0.1	localhost
::1		localhost

2001:bf7:820:901::1	stuttgarter-core.olsr	# myself
2001:bf7:830:82::1	f2a-finow-sued-2ghz.olsr	# 2001:bf7:830:82::1
2001:bf7:830:81::1	mid1.f2a-finow-sued-2ghz.olsr	# 2001:bf7:830:82::1 (mid #1)
2001:bf7:820:950::1	stuttgarter-rhnk.olsr	# 2001:bf7:820:950::1
...
```

bittorf, 2015-11-18:
ok, is it guaranteed that an IPv6-address has "*[0-9]::[0-9]*" in it?

bittorf, 2015-11-18:
can you please try if this works:
http://intercity-vpn.de/files/openwrt/neigh.sh

bittorf, 2015-11-20:
any progress?

booo, 2015-12-31:
We fixed this upstream. This will be part of the new firmware once we upgrade the dependencies. I'm closing this for now.

closed 2015-12-31 by booo