freifunk-berlin/firmware #309: add community-profile cottbus

ffcb, 2015-12-11:
Im Master von OpenWrt wurde der Commit https://github.com/openwrt/luci/commit/3bae3c7f5e8c383e481667fcfeae7765a1dacc75 mit unserem Community-Profil gemerged. Könnt ihr dieses Profil bitte auch in kathleen integrieren? Danke

booo, 2015-12-24:
I will try to take look at this.

FreifunkUFO, 2016-01-08:
gibt es schon erste Erfahrungen mit nicht-Berliner Profilen?

SvenRoederer, 2016-01-08:
in build https://github.com/SvenRoederer/openwrt-luci/tree/for-15.05_freifunk-berlin_unstable-SAm0815 ist es schin drin. 
ich will das alles stückweise richtung kathleem 0.2.0 pushen

ffcb, 2016-01-09:
ich habe auf einem nexx wt3020f mit der aktuellen weekly-alpha von kathleen 0.2.0 das profil nach dem flashen und einem reset per ssh manuell eingespielt. nach dem durchlauf mit dem freifunkassistenten hat sich der nexx sofort mit dem vorhandenen mesh verbunden

SvenRoederer, 2016-04-03:
these profiles are per default always included in all builds of branch "unstable_SAm0815" since 12. Jan 2016 (https://github.com/SvenRoederer/openwrt-luci/commit/8ecf17070f839a37a65e6b415a9efee77a4599f4). Check version.txt in root-folder of your firmware-build

SvenRoederer, 2016-04-26:
closed by https://github.com/freifunk-berlin/firmware/commit/3be098e246a3d82121c05bbdc213d2dd06b70f11

closed 2016-04-26 by SvenRoederer