freifunk-berlin/firmware #310: upsetram recent community-profile

SvenRoederer, 2015-12-18:
opened PR (openwrt/luci#606) to get our profiles updated in luci. When this is done we can remove 
- patches/013
- patches/015

sarumpaet, 2015-12-19:
There is already a branch and issue for this
https://github.com/freifunk-berlin/firmware/issues/245
https://github.com/freifunk-berlin/luci/commits/freifunk_berlin_updates

In your branch, you deleted the OLSRd defaults, which probably are important:
> option 'LinkQualityAlgorithm' 'etx_ff'

SvenRoederer, 2015-12-20:
> option 'LinkQualityAlgorithm' 'etx_ff'

gets also removed by the original patch used for current firmware (https://github.com/freifunk-berlin/firmware/commit/4558f8c). This patch is only intended to push the current settings.
As https://github.com/openwrt/luci/pull/606 is merged I checked both files after doing "make patch" and  diff-tool doesn't find any difference.

sarumpaet, 2015-12-20:
The Kathleen wizard doesn't implement the full profiles "standard". Some settings normally included in Freifunk/LuCI profiles are managed separately in Kathleen (I don't know why this was done, and I don't know why it wasn't properly documented, but it's that way at the moment). For example, OLSRd settings can be found in https://github.com/freifunk-berlin/firmware-packages/blob/master/utils/freifunk-berlin-olsrd-defaults/uci-defaults/freifunk-berlin-olsrd-defaults . By removing them from the generic LuCI profile, you broke the Berlin profile for people not using Kathleen.

pmelange, 2018-11-04:
All referred to issues and PR's have either been closed or merged.  I am closing this.  If this is still an issue, please reopen.

closed 2018-11-04 by pmelange