freifunk-berlin/firmware #311: DHCP-range not using full network-size

SvenRoederer, 2015-12-19:
I configured a /24-prefix on the DHCP-interface, but dnsmasq only uses 150 addresses

> daemon.info dnsmasq-dhcp[1586]: DHCP, IP range 10.230.192.100 -- 10.230.192.249, lease time 5m
> 
> config interface 'dhcp'
>        option proto 'static'
>        option dns '85.214.20.141 213.73.91.35 194.150.168.168 2001:4ce8::53 2001:910:800::12'
>        option ifname 'eth0'
>        option netmask '255.255.255.0'
>        option ip6assign '64'
>        option ipaddr '10.230.192.1'
>        option type 'bridge'
>
> ip addr show dev br-dhcp
>  inet 10.230.192.1/24 brd 10.230.192.255 scope global br-dhcp

booo, 2015-12-19:
Please post the output of /etc/config/dhcp and show us the content of /tmp/etc/dnsmasq.conf.

SvenRoederer, 2015-12-19:
<pre>
root@berlin-bhf-sxf01:/# cat /etc/config/dhcp 

config dnsmasq
        option domainneeded '1'
        option boguspriv '1'
        option filterwin2k '0'
        option localise_queries '1'
        option rebind_protection '1'
        option rebind_localhost '1'
        option local '/lan/'
        option domain 'lan'
        option expandhosts '1'
        option nonegcache '0'
        option authoritative '1'
        option readethers '1'
        option leasefile '/tmp/dhcp.leases'
        option resolvfile '/tmp/resolv.conf.auto'

config dhcp 'lan'
        option interface 'lan'
        option start '100'
        option limit '150'
        option leasetime '12h'
        option dhcpv6 'server'
        option ra 'server'

config dhcp 'wan'
        option interface 'wan'
        option ignore '1'

config domain 'frei_funk'
        option name 'frei.funk'
        option ip '10.230.192.1'

config domain 'frei_funk_ipv6'
        option name 'frei.funk'
        option ip 'fdf2:1da4:48d7::1'

config odhcpd 'odhcpd'
        option maindhcp '0'
        option leasefile '/tmp/hosts/odhcpd'
        option leasetrigger '/usr/sbin/odhcpd-update'

config dhcp 'dhcp'
        option leasetime '5m'
        option force '1'
        option interface 'dhcp'
        list dhcp_option '119,olsr'
        option dhcpv6 'server'
        option ra 'server'
        option limit '250'
        option ra_management '1'
        option start '3'

root@berlin-bhf-sxf01:/# cat /tmp/etc/dnsmasq.conf 
# auto-generated config file from /etc/config/dhcp
conf-file=/etc/dnsmasq.conf
dhcp-authoritative
domain-needed
localise-queries
read-ethers
bogus-priv
expand-hosts
domain=lan
server=/lan/
dhcp-leasefile=/tmp/dhcp.leases
resolv-file=/tmp/resolv.conf.auto
addn-hosts=/tmp/hosts
conf-dir=/tmp/dnsmasq.d
stop-dns-rebind
rebind-localhost-ok
dhcp-broadcast=tag:needs-broadcast
dhcp-range=dhcp,10.230.192.3,10.230.192.252,255.255.255.0,5m
dhcp-option=dhcp,119,olsr
</pre>

booo, 2015-12-19:
`dhcp-range=dhcp,10.230.192.3,10.230.192.252,255.255.255.0,5m` looks quite ok to me.

What ranges do you get as client?

SvenRoederer, 2015-12-19:
btw. when changing uci dhcp.dhcp.start & uci dhcp.dhcp.limit these values get used

SvenRoederer, 2015-12-19:
@booo initially posted the already modified config. changed the 3 comment just before you answered to the original values I had

booo, 2015-12-19:
@SvenRoederer please read the documentation: https://wiki.openwrt.org/doc/uci/dhcp If this still does not make sense to you please let me know.

SvenRoederer, 2015-12-21:
Just rechecked on a plain install:
> /tmp/etc/dnsmasq.conf
> dhcp-range=dhcp,10.230.194.100,10.230.194.249,255.255.255.0,5m
> 
> /etc/config/dhcp
> config dhcp 'dhcp'
>         option leasetime '5m'
>         option ignore '0'
>         option force '1'
>         option interface 'dhcp'
>         list dhcp_option '119,olsr'
>         option dhcpv6 'server'
>         option ra 'server'

and syslog gives:
> DHCP, IP range 10.230.194.100 -- 10.230.194.249, lease time 5m

As uci dhcp.dhcp.start & dhcp.dhcp.limit are undefined the defaults of 100 / 150 are used. 
But should these values not defined to give out as much DHCP-leases as possible for the avail address-range?

booo, 2015-12-22:
Yes, we should compute the right values for start and limit based on the subnet the user entered.

If we don't use start and limit the init script computes a range based on the subnet and the defaults (100, 150). The result depends on the subnet size and makes in most cases no sense.

This is a quit stupid bug :/

booo, 2015-12-25:
I think we can compute start and limit like this:

```lua
local limit = math.pow(2, 32 - prefix) - 2                                                                                             
local start = 2
```

booo, 2015-12-31:
@SvenRoederer Can you test the final patch again? I tested it here and it seemed to work. I would appreciate a test of the migration script too. Works here.

SvenRoederer, 2016-01-01:
for some reason it fails to lookup the ${subnet} during migration.
for debug I did
<pre>
log() {
  logger -s -t freifunk-berlin-migration $@
  echo >>/tmp/uci-up.log $@
}

fix_dhcp_start_limit() {
  log "mig dhcp_start_limit"
  # only set start and limit if we have a dhcp section
  if (uci show dhcp.dhcp); then
    # only alter start and limit if not set by the user
    if ! (uci get dhcp.dhcp.start || uci get dhcp.dhcp.limit); then
      logger 'starting real work'
      local subnet
      local prefix
      # get network in cidr format
      network_get_subnet "subnet" "dhcp"
      log "got subnet: ${subnet}"
      # extract prefix with basename command
      prefix=$(basename ${subnet})
      # compute limit (2^(32-prefix)-3) with arithmetic evaluation
      limit=$((2**(32-${prefix})-3))
      uci set dhcp.dhcp.start=2
      uci set dhcp.dhcp.limit=${limit}
      log after uci-set
    fi
  fi
  log "finishing dhcp_limit_fix"
}
</pre>

and got following output in /tmp/uci-up.log
> migration starting ...
Migrating from 0.1.2 to 0.2.0-alpha.gitab84fe2.
to 0.2.0
mig dhcp_start_limit
got subnet:

I feel that this is caused by still disabled network-interfaces

SvenRoederer, 2016-01-01:
I checked the interfaces during runtime of uci-default scripts with "ip" --> interfaces are not setup

SvenRoederer, 2016-01-02:
- fixed this in https://github.com/freifunk-berlin/firmware-packages/pull/67
- checked for function on real  system

booo, 2016-02-15:
The pull request is merged. Closing this.

closed 2016-02-15 by booo